dgiese/dustcloud #1: Where to get new firmwares from

jvitkauskas, 2017-12-27:
Is there a way to download latest firmware without risking it installing on the device?

dgiese, 2017-12-27:
Hi, technically there is, i will upload some infos later. I uploaded the Version 3094 now. However i did not test it yet in terms of rooting.

jvitkauskas, 2017-12-27:
Thanks. I cannot check the app as I don't have the device yet. Also, it would great if you could add a link to http://dontvacuum.me/ in README.md if you intend to post something more there.

Flole998, 2017-12-29:
3094 and even 3124 work. No issues when rooting them. I'm also curious about how you know which is the latest firmware and where it can be downloaded.

linuxlurak, 2017-12-30:
@Flole998 could you upload  	v11_003132 to your fork? would be super nice!

Flole998, 2017-12-30:
3132 is more than 100 MB, so it's not possible. You can however try my App if you just want to install it without any modifications

linuxlurak, 2018-01-01:
Chek this one:
https://dustcloud.seemoo.de/public/xiaomi.vacuum.gen1/original/encrypted/

linuxlurak, 2018-01-01:
question still persists further on: how did you grad those files? is there a technique to predict the links?

dgiese, 2018-01-04:
You can read them in the dustcloud or in the logs of the vacuum.

closed 2018-01-21 by dgiese