dgiese/dustcloud #2: Instructions on how to root with built firmware

akeilox, 2017-12-28:
I came across this link via http://www.robotreviews.com/chat/viewtopic.php?f=6&t=19479&start=780
therefore i didnt attend the session, merely trying to get an understanding on how one builds a firmware with the .sh script and most importantly how one can carry the actually rooting to the robot. I know of the micro USB port an Mi App. Assuming there is a certain process involved with micro usb port?

I have downloaded the repository but cant seem to find the instructions on how to apply the built firmware to the Mi vacuum v1. 
When i take a quick look at the imagebuilder.sh i see the "authorized_keys" file, is this something generate or extracted from somewhere as I cant seem to locate the file.

Would you kindly share, or better include it as part of the repository, on how to use these tools and carry the actual rooting and fallback to original firmware (if things go wrong)?

thanks for the excellent work and sharing with the community!

Henne78, 2017-12-28:
You can generate your ssh-keypair with public- and private key using PuttyGen. Did you read the instructions at:

[https://github.com/dgiese/dustcloud/blob/master/UPDATE-howto.md](url)

Henne78, 2017-12-28:
Could someone tell me which command I have to use in Python to start the imagebuilder script after putting the 3 files (authorized_keys) inside the folder of the imagebuilder.sh? Do I have to put the ssh-keypair files, which I saved before, inside the folder with the imagebuilder.sh or just only the authorized_keys with the public key inside?

dgiese, 2017-12-28:
you just start the imagebuilder.sh with bash... for building the image you do not need python.

frostworx, 2017-12-28:
shouldn't 
	cp authorized_keys ./root/.ssh/
be 
	cp ../authorized_keys ./root/.ssh/
pwd is image.
great work! can't wait to play with it! thanks a lot!

dgiese, 2018-01-04:
Should be solved now.

closed 2018-01-04 by dgiese