dgiese/dustcloud #4: Question: What exactly (eg wfif credentials) is uploaded to xiaomi servers?

linuxlurak, 2017-12-28:
Thank you for your speak yesterday at 34c3! You did tremendous work! Your efforts will affect many other projects in a positive way.

Something that was not completely clear to me was the point you talked about "data" that is uploaded to the xiaomi servers. In your slide data consists log files and this logfiles in turn contain credentials. 

I'am referring to this slide:

[image]

Do the xiaomi servers grab credentials?

dgiese, 2017-12-28:
take a look into /mnt/data/rockrobo/rrlog/ and answer the question for yourself ;)

linuxlurak, 2017-12-28:
thank you for treating me as a hacker ;D I'm a honored rookie! I did not root my staubi at the moment. Will be done soon.

Right now I mounted the disk revison 11.003077 (v11_003077.pkg.tar.gz) and looked at `/opt/rockrobo/rrlog/rrlog.conf` where I found:

> MAX_TAR_FILE_SIZE=734003200
> #UPLOAD_METHOD   0:NO_UPLOAD    1:FTP    2:FDS
> UPLOAD_METHOD=2
> #UPLOAD_URL=ftp://ftpuser:passwd@192.168.176.157
> UPLOAD_URL=https://awsbj0.fds.api.xiaomi.com
> #UPLOAD_URL=http://files.fds.api.xiaomi.com
> LOG_FLASH_PATH=/mnt/data/rockrobo/rrlog
> LOG_COUNTER_FILE=/mnt/reserve/counter
> BKBOX_COUNTER_FILE=/mnt/reserve/bkcounter
> #MAX_UPLOAD_SPEED is in unit of bytes/s
> MAX_UPLOAD_SPEED=51200
> ONLY_UPLOAD_ONDOCK=1
> #LEVEL_ERROR = 3U #LEVEL_WARN = 4U #LEVEL_INFO = 6U #LEVEL_DEBUG = 7U #LEVEl_VERBOSE = 8U
> LOG_LEVEL=6
> **#UPLOAD_EN_DEFAULT**      0: will not upload log  1: will upload bkbox by default   3:will upload bkbox and EVENTASK 7: upload bkbox and all logs

#UPLOAD_EN_DEFAULT  is commented out...

and `/opt/rockrobo/rrlog/rrlogmt.conf`:

> MAX_TAR_FILE_SIZE=734003200
> #UPLOAD_METHOD   0:NO_UPLOAD    1:FTP    2:FDS
> UPLOAD_METHOD=2
> #UPLOAD_URL=ftp://ftpuser:passwd@192.168.176.157
> UPLOAD_URL=https://awsbj0-files.fds.api.xiaomi.com
> #UPLOAD_URL=http://files.fds.api.xiaomi.com
> LOG_FLASH_PATH=/mnt/data/rockrobo/rrlog
> LOG_COUNTER_FILE=/mnt/reserve/counter
> BKBOX_COUNTER_FILE=/mnt/reserve/bkcounter
> #MAX_UPLOAD_SPEED is in unit of bytes/s
> MAX_UPLOAD_SPEED=51200
> ONLY_UPLOAD_ONDOCK=1
> #LEVEL_ERROR = 3U #LEVEL_WARN = 4U #LEVEL_INFO = 6U #LEVEL_DEBUG = 7U #LEVEl_VERBOSE = 8U
> LOG_LEVEL=8
> #UPLOAD_EN_DEFAULT      0: will not upload log  1: will upload bkbox by default   3:will upload bkbox and EVENTASK 7: upload bkbox and all logs
> UPLOAD_EN_DEFAULT=0

Here UPLOAD_EN_DEFAULT is set to 0. Looks a little better.

Have to check whats running on the living robovac.

cheers!

dgiese, 2017-12-28:
Hi,
so there is some cloud command to toggle the log upload status. While one of my vacuums was connected to the cloud, i noticed the upload of some files to their servers. However i have that root access already for a few months and work with an older firmware version. The newer versions seems to be less talkative and create less verbose logs. If anyone want to play around with the log uploads, here are the commands for that:

    else if ( !strcmp(v5, "disable_log_upload") )
    {
      result = AppProxyServer::HandleDisableLogUpload(v3, v2);
    }
    else if ( !strcmp(v5, "enable_log_upload") )
    {
      result = AppProxyServer::HandleEnableLogUpload(v3, v2);
    }
    else if ( !strcmp(v5, "get_log_upload_status") )
    {
      result = AppProxyServer::HandleGetLogUploadStatus(v3, v2);
}

closed 2018-01-21 by dgiese

JamesBream, 2018-02-24:
@dgiese So how exactly are log uploads disabled? I've just bought one of these and stumbled across this repo. Not sure I like the idea of the vac uploading wifi credentials to the cloud if I can help it.