dgiese/dustcloud #6: Idea:

SvenGWK, 2017-12-28:
Great work!  I guess now it is also possible to extend the functionality of the "old" robot?  I think there, for example  to the subdivision of the floor plan of an apartment into individual rooms.  This would have the advantage that you do not have to  stick the ugly adhesive strips ("Virtual walls") on the floor and no longer has to drag the robot into the individual rooms.
It would be great if you could just call the robot into the living room if you need it right there.
The successor of the our robot has afaik such a function, I suspect this will also be exclusive for this.

matphillips, 2017-12-28:
Glad to see there is interest in this project, great work by the devs!
Bear in mind this is basically a way to root the device - what you do with it from there is anyones guess.
This project could lead the way to a custom and enhanced firmware, but in and of itself, it's just a way to gain root access to the device.

For the concept you mention here @SvenGWK I would have to say no, it is unlikely. The map is generated at runtime as the robot learns its environment, there's no stored work-map, and so nothing to mark demarkation points on in advance. That said, anything is possible if someone creates a sufficiently advance firmware in the future.

I'll be happy to plug it into Homebridge as an on/off device, and have it emulate a security camera to send its map images back, which will be sufficient functionality while having it isolated from cloud servers. I will be investigating the implementation of this.

dgiese, 2017-12-28:
I *strongly* believe that you could port back functions from Gen2 to Gen1. At the moment i do not have a Gen2 myself, therefore i cannot investigate myself on that topic.

Flole998, 2017-12-29:
I tried to run the v2 software on the v1: As it's missing 2 sensors it causes problems and it won't even clean because of the sensor issue.

dgiese, 2017-12-29:
You mean v2 firmware or v2 app software?

Flole998, 2017-12-29:
V2 firmware. I replaced /opt and then restored the firmware for the stm32 so it doesn't try to flash that thing as I wasn't sure if the pinout is the same

dgiese, 2017-12-29:
Hi, which version number of v2?
You should not simply restore the firmware to the stm32 as the layout might be different. If something gets broken with the stm, then your whole vacuum is broken, as it controls the power to the R16.

Flole998, 2017-12-29:
I know, thats why i replaced the v2 firmware folder with the one from v1 so it doesn't do that.

I used the latest v2 firmware, it was 001168.

Markus1995, 2017-12-29:
Maybe you can upload the Firmware from v2.

Flole998, 2017-12-29:
On the STM? Maybe, but most likely it'll brick the robot.

Markus1995, 2017-12-29:
No, i mean on GitHub.

Flole998, 2017-12-29:
Assuming that you mean the ARM professor: that will, if applied unchanged, flash the stm. That would result in a brick. I've tried with a modified version that doesn't mess around with the STM but it is complaining about internal errors when it's supposed to clean. Most likely because it's missing 2 sensors.

dgiese, 2017-12-30:
Hi, the firmware is bigger than 100mbyte, therefore i cannot upload it to Github :/

buzzzzer, 2018-01-10:
Can someone put the gen2  irmware on any hosting ? 

I successfully changed the sounds to my own with mirobo dnld_install_sound on my gen2 robot.
But without firmware  I do not get root

closed 2018-02-18 by dgiese