dgiese/dustcloud #9: python-miio instead of python-mirobo?

frostworx, 2017-12-28:
hi,
are you planning to support
the active https://github.com/rytilahti/python-miio
project instead of the old https://github.com/ultrara1n/python-mirobo old version?
or does it *just work*?

dgiese, 2017-12-28:
Hi,
to be honest, i dont know. When i developed the initial root many months ago, there was no python-miio. I still use an old python-mirobo version. If python-miio have the same featureset, then it should work just so.
It would be great if someone could test it.

frostworx, 2017-12-28:
thanks for the quick answer! I'll tell you as soon as I know.

frostworx, 2017-12-28:
hi again!
just tested with =dev-python/python-miio-0.3.3 using python 3.5 on gentoo:
 mirobo discover --handshake true # worked fine
 mirobo --ip=192.168.8.1 --token=#Token_from_above# status # worked fine
 mirobo ... raw_command... # started fine. status was "Updating" as expected, and after some minutes the robot confirms this with sysupd_start.wav, just a second after it stops with sysupd_failed.wav though :(
tried both with v11_003077.pkg and v11_003094.pkg running a quickly installed nginx listening on 0.0.0.0
download worked fine from a different machine.
currently installed version already is on 3077 - maybe a region problem?
not sure if it is python-miio's fault - at least it doesn't drop any errors. 
if there's anything else I could test I'd be glad to help.

Flole998, 2017-12-28:
I used python-miio and it is working perfectly. I still have to create a PR for the update method I implemented there, or I'll just drop the few lines here so one of you guys can do it if you want. 

What you are describing sounds like an MD5 checksum error, you should doublecheck that. Also make sure the robot is in the charging station. Don't ask me why, I don't understand that either. Just do it ;)

frostworx, 2017-12-28:
ok, nice. thx for testing! the robot was in the charger. I already thought it might not even find the provided md5 file. assuming it is read from pwd, I also put it into htdocs and provided a relative path but both didn't help. trying again tomorrow :)

Flole998, 2017-12-28:
Don't provide the path but put the md5sum directly in there.

frostworx, 2017-12-28:
heh, did exactly that and just wanted to report back, that "file_md5" is misleading.
everything worked fine. thanks for the help!

closed 2017-12-28 by frostworx