dgiese/dustcloud #11: Test Application

EliasKotlyar, 2017-12-29:
Hello everyone,

I found a really nice testing application on the robot, which has a lot of funny options. You can find it under: /opt/rockrobo/firmware/uart_test

It allows the following:
```
root@rockrobo:/mnt/data/rockrobo/sounds# uart_test-3.2.1-20170620 -h
[ 3358027]<N>uart_test-3.2.1-20170620 VERSION:3.2.1
[ 3358028]<N>Usage:
[ 3358028]<N>Options:
[ 3358028]<N> -v <v1> <v2> <v3>     linespeed,angular velocity,linespeed when a turn
[ 3358028]<N> -f <speed>            set fan speed 0~100
[ 3358028]<N> -b <speed>            set brush speed 0~100
[ 3358028]<N> -s <speed>            set sweep speed 0~100
[ 3358028]<N> -e <wav_file>         play sound
[ 3358028]<N> -g                    get mcu status
[ 3358028]<N> -m <nisfb>            set mcu into the mode
[ 3358028]<N>     n                    set mcu into nomal mode (default)
[ 3358029]<N>     i                    set mcu into idle mode
[ 3358029]<N>     s                    set mcu into shutdown mode
[ 3358029]<N>     r                    set mcu reset
[ 3358029]<N> -p <adinweclg>        print mcu debug log
[ 3358029]<N>     a                    print info and above log
[ 3358029]<N>     d                    print debug and above log
[ 3358029]<N>     i                    print info and above log
[ 3358029]<N>     n                    print notice and above log
[ 3358029]<N>     w                    print Warning and above log
[ 3358029]<N>     e                    print erro and above log
[ 3358029]<N>     c                    print crit and above log
[ 3358030]<N>     l                    print alert and above log
[ 3358030]<N>     g                    print emerg log
[ 3358030]<N>     s                    print uart raw data
[ 3358030]<N> -u <mcu_firmware>     upgrade_mcu fw
[ 3358030]<N> -U                    Auto Upgrade Mcu
[ 3358030]<N> -l                    test led command
[ 3358030]<N> -t <abcd...z>         test case for debug
[ 3358030]<N>     AT command           facotory or bit command
[ 3358030]<N> -d <abcd...z>         debug the report result from mcu
[ 3358030]<N>     a                    print all mcu status
[ 3358030]<N>     d                    print dock vol if mcu report
[ 3358030]<N>     s                    print all switch status(bumper,drop,dock,bat_info,key,dustin)
[ 3358031]<N>     c                    print wheel current
[ 3358031]<N>     l                    print acc_gyro_quat_odo_time
[ 3358031]<N>     w                    print wall
[ 3358031]<N>     u                    print ultra
[ 3358031]<N>     t                    print mcu ts offset
[ 3358031]<N> -w <lds_spd>         set lds spd(200~400)
[ 3358031]<N> -r                   test random
[ 3358031]<Help>
support key contrl when test move runtime:
[ 3358031]<Help>    h        print help
[ 3358031]<Help>    w        go front
[ 3358032]<Help>    s        go back
[ 3358032]<Help>    a        turn left
[ 3358032]<Help>    d        trun right
[ 3358032]<Help>    z        force stop move
[ 3358032]<Help>    x        force/normal mode
[ 3358032]<Help>    UP       force go front
[ 3358032]<Help>    LEFT     force turn left
[ 3358032]<Help>    RIGHT    force turn rigth
[ 3358032]<Help>    DOWN     force go back
[ 3358032]<Help>    b        disable/enable mcu_heartbeat
[ 3358033]<Help>    f        start fan sweep and brush
[ 3358033]<Help>    t        start fan,set sweep and brush inversion
[ 3358033]<Help>    g        stop fan sweep and brush
[ 3358033]<Help>    o        open all switch
[ 3358033]<Help>    c        close all switch
[ 3358033]<Help>    e        set mcu print debug log
[ 3358033]<Help>    l        set mcu disable/enable charging
[ 3358033]<Help>
[ 3358033]<N>
support key contrl to get mcu status:
[ 3358033]<N>    h        print help info
[ 3358034]<N>    a        print dustin status
[ 3358034]<N>    b        print bumper status
[ 3358034]<N>    c        print cliff status
[ 3358034]<N>    d        print dock status
[ 3358034]<N>    e        print drop status
[ 3358034]<N>    f        print fan info
[ 3358034]<N>    i        print TestInfo status
[ 3358034]<N>    j        print CaliInfo
[ 3358034]<N>    k        print secret code
[ 3358034]<N>    l        print wheel info
[ 3358034]<N>    m        print main brush info
[ 3358035]<N>    n        print SN and PN
[ 3358035]<N>    o        print odo and gyro status
[ 3358035]<N>    p        print power status
[ 3358035]<N>    r        print Dev Info
[ 3358035]<N>    s        print sweep info
[ 3358035]<N>    t        print mcu ts offset

[ 3358035]<N>    u        print ultra distance
[ 3358035]<N>    v        print mcu version and sys_mode
[ 3358035]<N>    w        print wall distance
[ 3358035]<N>    x        print bootloader Version
[ 3358035]<N>    z        print mcu RTC
[ 3358036]<N>    Z        set cur ap time to mcu RTC
[ 3358036]<Help>
 support key contrl when test led runtime:
[ 3358036]<Help>    h        print help info
[ 3358036]<Help>    l        change led light level
[ 3358036]<Help>    s        sent led command
[ 3358036]<Help>    c        change set mode(cur or rstr)
[ 3358036]<Help>
[ 3358036]<Help>    n        none
[ 3358036]<Help>    r        red color
[ 3358036]<Help>    o        orange color
[ 3358037]<Help>    w        white color
[ 3358037]<Help>
[ 3358037]<Help>    1        turn on
[ 3358037]<Help>    2        turn off
[ 3358037]<Help>    3        blink
[ 3358037]<Help>    4        fastblink
[ 3358037]<Help>    5        breath
[ 3358037]<Help>    6        deepbreath
[ 3358037]<Help>
[ 3358037]<Help>    0        set repeat = 0
[ 3358037]<Help>    8        set repeat = 4
[ 3358038]<Help>    9        set repeat = -1
[ 3358038]<Help>

```

dgiese, 2017-12-29:
Hi,
yes. However you have to kill the player or RoboController process (did not remember which one). The watchdog will restart it after some time. You can disable the watchdog if you comment it out in /opt/rockrobo/watchdog/ProcessList.conf or /opt/rockrobo/watchdog/ProcessListMT.conf .

EliasKotlyar, 2017-12-29:
Thanks for the information. I deactivated both processes by commenting them in the "/opt/rockrobo/watchdog/ProcessList.conf". After that, you can drive the robot with the following command:
```
uart_test -v 1000 1000 1000
```

Also you can setup the LEDs using "-l" option.

YoranSys, 2017-12-31:
Hi very nice hack @dgiese and @EliasKotlyar, 
for playing radio we can do that : 
``` apt-get install sox wget libsox-fmt-mp3```
```sox http://chai5she.cdn.dvmr.fr/fipnantes-midfi.mp3 -t wav - | /opt/rockrobo/firmware/uart_test -e /dev/stdin```
The sound is pretty good

closed 2018-07-19 by dgiese