dgiese/dustcloud #12: question: search token

frostworx, 2017-12-29:
hi,
rooting my vacuum worked fine, but of course my previously used token to remote control it using python-miio was replaced by a random new one. the new created token is no longer valid though after having joined my iot network, so I wonder if anybody already found out where it could be extracted/generated directly on the rockrobo system, so controlling the vacuum via python-miio works again. already searched for it but haven't found anything that looks like a 32 byte token yet anywhere.

Flole998, 2017-12-29:
/mnt/data/miio/device.token is storing it, but it's not stored in plaintext. Maybe md5 of that file or something. That would also generate a random token if it doesn't exist.

frostworx, 2017-12-29:
thank you! I've seen that file, but I thought the "Device Token" description in Interesting-files.md means, that this is the unique device id, which doesn't change.
I took the "old classic way" to receive a new valid token (nice description here: https://github.com/jghaanstra/com.xiaomi-miio/blob/master/docs/obtain_token_mirobot_new.md).
It was some extra work to open all firewall and dns blocks for the (damn!) android tablet and the robot, so it would be nice if the token could be extracted otherwise.
In  case I find out something useful I'll report back.

bademux, 2017-12-29:
Will ```mirobo discover --handshake true``` work for token or network is needed anyway?
I want to disable any communication with the cloud on my Vacuum
```
	echo "0.0.0.0       awsbj0-files.fds.api.xiaomi.com" >> ./etc/hosts
	echo "0.0.0.0       awsbj0.fds.api.xiaomi.com" >> ./etc/hosts
	echo "0.0.0.0       ott.io.mi.com" >> ./etc/hosts
	echo "0.0.0.0       ot.io.mi.com" >> ./etc/hosts
```

Flole998, 2017-12-29:
Should still work, however the token will change often because it resets it after some time if it's not connected to the cloud. Or at least if it hasn't been registered there before as part of the setup process.

dgiese, 2017-12-29:
Hi,
@bademux: Token is issued by miio_client, so blocking the cloud have no effekt.
@frostworx: convert the token to hex, e.g. Jm5RKQ05IABCDEFG -> 4a6d35524b5130354941424344454647 (lowercase!)
Dennis
(Sorry for late reply, had to get some sleep at the 34C3)

frostworx, 2017-12-29:
yay, thank you very much @dgiese!
for blocking the cloud I'd also add "sg.ott.io.mi.com" to the list.
until yesterday I left open port 8053 for two xiaomi servers open on my firewall, which allowed me to keep my token and close evertything else. I guess I don't need that anymore now :)

bademux, 2017-12-29:
Sorry for stupid question, but maybe we can allow only local network on the Vacuum itself?

dgiese, 2017-12-29:
The servers depend on which server your mi home app works. I always use mainland china to get all the devices. For different settings the servers will be different.

frostworx, 2017-12-29:
printf $(cat /mnt/data/miio/device.token) | xxd -p
works fine, dgiese. thanks again! :}

Flole998, 2017-12-29:
`iptables -A OUTPUT -d 192.168.0.0/24 -j ACCEPT
iptables -A OUTPUT -d 127.0.0.0/8 -j ACCEPT
iptables -A OUTPUT -j DROP`

This will block everything except for 192.168.0.0/24 and local connections. Use with caution, you can't install voice packs or firmware from remote servers anymore. Also apt update won't work anymore.

sedrubal, 2018-03-13:
For me the key in `/mnt/data/miio/device.token` does not seem to be the correct token.
But sending the magic package to the miio_client reveals the correct token.
Here is another method to get it:
- start wireshark or tcpdump (on your pc or on the robot)
- `echo -ne '\x21\x31\x00\x20\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff' | nc -u 192.168.8.1 54321`
- extract the last 16 bytes/characters and convert them to a (32 characters) hexadecimal string (using `xxd -p`)

Patattack666, 2018-05-18:
This seems to work for latest version (presuming you have a rooted android device): 

- Delete the app -> reinstall -> log in. 

- Once logged in, use adb to access the rooted device through USB cable. In: 
`/data/data/com.xiaomi.smarthome/cache/smrc4-cache`

- print the content of each file in this folder (cat *)

this will give you the cached decrypted 'RC4 encrypted' settings which are fetched through the various APIs. This will print various JSON objects, whereas you'll find the Token of your device. 

// Pat

closed 2018-07-19 by dgiese