dgiese/dustcloud #14: Player unable to gathet map data

Flole998, 2017-12-29:
As seen in the presentation, you were able to use playerv or playernav to get the map data. When i request map:0 using playerv I'm only getting timeouts, requesting laser:3 works fine. What am i doing wrong? How did you get that data?

dgiese, 2017-12-29:
Hi,
so first you have to get rid of the iptables rules (e.g. by iptables -F), however this opens all ports to your local network!
I used robot-playerv -h 192.168.8.1 (robot-playerv is a package in ubuntu 14.04, so you do not need to compile it. 192.168.8.1 is the IP of the robot when it creates the open accesspoint. I did the pictures from laser:0. This gives you a live picture of the current laser measurement.
The complete maps were from /run/shm (see interesting-files.md). Or you can unpack them from the data partition (they are the ppm files compressed in the log files).

Flole998, 2017-12-29:
I am able to get the laser data, so firewall rules are fine. I am just not able to get map:0 and I thought that you were able to get that, but seems like you didn't try that,

dgiese, 2017-12-29:
Right, you cant get the map over playerv. I used the .ppm file for that. Xiaomi uses some weird format for the upload to their servers. There is some sample code in the dustcloud, which returns the map urls. I will upload the corresponding server part.

Flole998, 2017-12-29:
The format is actually quite simple, I have figured that out some time ago but I haven't yet implemented it in my App, as I am unable to get the data from the xiaomi cloud. The robot doesn't give the access token and the App uses some weird protocol to talk to their servers.

dgiese, 2017-12-29:
Look for dustcloud/mapupload. There is a sample file.

Flole998, 2017-12-29:
I also have plenty of those sample files availabe, as the xiaomi App writes the map url to logcat when it downloads the map, but that doesn't really help me as there is an access token included in the request. Without that token it won't give me the map. I need to figure out a way to get a valid token so I can get the map data.

dgiese, 2017-12-29:
Hi,
you do not need an access token if you upload it to your own server ;) There is a sample implementation for the server side. You need to adjust the URLs in the Dustcloud.

Flole998, 2017-12-29:
Yeah thats possible of course, but not wanted in this case ;) However, the player thing is not in issue, so this can be closed.

closed 2017-12-29 by Flole998