dgiese/dustcloud #15: Offline only configuration possible?

fvollmer, 2017-12-29:
Can I use the robot in a separated WiFi without internet?

I just installed a rooted `v11_003077.pkg` firmware on my new and never used robot. After this I could get status infos etc. via `mirobo`. To avoid the open WiFi I connected the robot to my newly created seperated wifi by executing

    mirobo --ip=192.168.8.1 --token=#my_token# configure_wifi #SSID# #password#

and then I was able to connect to the robot via SSH using the separated network. To get the new token I executed:

    printf $(cat /mnt/data/miio/device.token) | xxd -p

For some reason all mirobo commands fail now. E.g. the command

    mirobo --ip=192.168.4.199 --token=#new_token# status

just returns  `Unable to discover the device 192.168.4.199`. I tried to check if the robot receives anything with `tcpdump port 54321 -v` and according to the output

    tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 65535 bytes
    01:53:53.288888 IP (tos 0x0, ttl 64, id 10468, offset 0, flags [DF], proto UDP (17), length 60)
        192.168.4.196.46105 > 192.168.4.199.54321: UDP, length 32
    01:54:03.842597 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 60)
        192.168.4.199.54321 > 192.168.4.196.46105: UDP, length 32

the robot receives packages. Any ideas what is missing to communicate locally with the robot?

fvollmer, 2017-12-29:
Just noticed the following `tcpdump` output

    4 packets captured
    4 packets received by filter
    0 packets dropped by kernel
    4 packets dropped by interface

dgiese, 2017-12-29:
see page 15 in techinfo.pdf

fvollmer, 2017-12-29:
Thank you!
I missed this part of the documentation:
> APP-commands are not accepted, if Cloud-communication was not established

bademux, 2017-12-29:
@fvollmer that is exactly what I want to achieve: cut off comunication with cloud and controll Vacuum with OpenHab.
@dgiese can I ask you about "Cloud-communication was not established", can we create dummy CloudAPI to make Vacuum think it pings real cloud? Or it is more than just a ping?

dgiese, 2017-12-29:
For this i have the dustcloud. However you need to host it somewhere. For old version you could simply replay the first few requests of the vacuum and the vacuum changed its state. But i dont know if it still works. Alternatively you could write a small program which listens to localhost:54322 and forwards the stuff to OpenHab

jvitkauskas, 2017-12-30:
Why not hosting it on the device itself?

Markus1995, 2017-12-30:
This would be Really great.

jvitkauskas, 2017-12-30:
Just found this in the slides:

> Block if cloudserver ip part of local subnet
• DNS modification to local server does not work

Does that mean that cloud connection to localhost is blocked? If so, maybe creating another virtual network interface is possible?

Flole998, 2017-12-31:
No, it means that it's using HTTPS, so you would have to fake the certificate

fvollmer, 2018-01-05:
I created an open source implementation of `miio_client` in C. At the moment it just handling the discovery and the messages forwarding between the ports `54321` and `54322`  with all the encrypting and md5 stuff.
The implementation appears to be working fine without a cloud connection even when the robot is provisioned (tested with `python-miio`), but maybe someone can help me with the following questions:
 * How can I find out if the device is provisioned? (A client shouldn't be able to discover the real token if the device is provisioned)
 * What token should I send, after the device received a discover command and is provisioned? 
 * How can I get the device ID?
 * What should I do with the stamps?
 * If there are multiple clients on the UDP port, how should the message forwarding work? I'm currently sending all answers to the last UDP address.

I will create a pull request at the weekend, after some code cleanup.

dgiese, 2018-01-05:
@fvollmer: Can you write me a mail? Maybe we can test your code against my cloud implementation or share ideas, without bothering the other people here.

1) If a wifi.conf file exists, then the vacuum is provisioned
2) you get the token usually by miio_client_helper_nomqtt.sh
3) is also told by miio_client_helper_nomqtt.sh
4) you mean the epoch?
5) broadcasting,  i guess.

dgiese, 2018-01-05:
Btw: i dont know if it makes sense to reimplement the miio_client completely. I had 2 ideas, for which i did not have time yet (and the knowledge for good socket, multi-client programming): 
1. Proxy all plaintext messages between AppProxy+Helper and Miio_client
2. Implement open Interface instead of Miio_client (e.g. for accepting plaintext json from FHEM/Home Assistant/etc. or providing a HTTPS service, ...)
-> The beauty of solution #1 would be, that you dont break functionality of the App but can still control the commands locally, providing an independant interface to the vacuum
-> Solution #2 is similar to one, but without all the Xiaomi stuff.

Flole998, 2018-01-05:
@dgiese for 2 it shouldn't send the token, as it's already provisioned. You should send a token full with f's, so fffffffff......

stamps are evaluated and need to match. theres a 1 second tolerance but otherwise they need to match.

You need to make sure that you keep track of which client requested what and route the answers accordingly.

timmmmmey, 2018-01-11:
@fvollmer  I would love to try this out!

devkid, 2018-01-21:
Would it be possible to remove most of the services on the device (shown on "Communication relations" in `techinfo.pdf`) and talk directly to the `player` service? If yes, what implications would that have? I guess not being able to use the app anymore is just one of them.

Flole998, 2018-01-21:
It depends on what you implement, everything you don't implement is a limitation. It's a lot of work to do so to even get it usable somehow as you need to figure out the commands send by the Robocontroller to the motorcontroller and the sensor stuff and so on

devkid, 2018-01-21:
Ok, then I probably misinterpreted what this `player` service is doing. I though it already handles all the mechanical and sensoric controlling? Did anyone find out what service is exactly responsible for what?

Flole998, 2018-01-21:
Robocontroller controls player and turns on and off the sensors and controls the led, player does navigation stuff and sends the commands to the motors. I think the Robocontroller needs to start them first though. Appproxy uploads maps, they are created by player, more specifically the navigation plugin. Miio client communicates with AppProxy. Did I forget a service?

merlink01, 2018-08-23:
Ok, we dont have iwconfig so I try another way.

merlink01, 2018-08-24:
Got it:
``
if [ "ping -c 1 your_router_IP" ]
then
  echo 'online'
else
 ifdown wlan0 > /dev/null 2>&1
 ifconfig wlan0 down > /dev/null 2>&1
 killall hostapd >/dev/null 2>&1
 iw mon.wlan0 del >/dev/null 2>&1
 create_ap --stop wlan0 > /dev/null 2>&1
 killall wpa_supplicant >/dev/null 2>&1
 killall dhclient >/dev/null 2>&1

 ifconfig wlan0 up >/dev/null 2>&1
 /sbin/wpa_supplicant -s -B -P /var/run/wpa_supplicant_1.wlan0.pid -i wlan0 -D nl80211,wext -c /mnt/data/mywlan/wpa_supplicant.conf -C /var/run/wpa_supplicant >/dev/null 2>&1
 dhclient wlan0 >/dev/null 2>&1
fi
``
now I have to create a cron job for this.

closed 2019-04-11 by fvollmer