dgiese/dustcloud #19: Alexa/Google Assistant Support?

Tuckeran5607, 2017-12-30:
Hey guys. Was curious if porting Support for GH or Echo devices would be possible after rooting.

I figure it should. Since all you have to do is send a signal to the device via the Internet to start cleaning...

Anyone smarter than me able to tell me about this?

tmechen, 2017-12-30:
you can actually achieve this even without rooting. you can add your xiaomi vacuum robot to your custom https://home-assistant.io/ installation. furthermore you can add your alexa etc to your hassio and create custom comands (regarding the vacuum)

Flole998, 2017-12-31:
Not only home assistant is possible, also OpenHAB and FHEM.

frostworx, 2017-12-31:
I prefer to (voice) control everything offline. currently snowboy launches my python-miio based scripts. contains a binary blob, but better than a wiretap. generally it is a matter of some basic scripts to control your devices (you need a commandline of course) and those scripts can be implemented everywhere (f.e. domoticz).

Tuckeran5607, 2017-12-31:
You guys have opened my eyes to a whole new area I didn't know was possible.

Anyone have a write-up or know where I can find one to get started with Alexa? I got my parents a Xiaomi Vac for Christmas, and I was under the impression it would work with the Singapore Server (Google Alexa and Xiaomi Vacuum.) 

But when I tried to set it up for them Christmas Day. The Singapore Server has been shut down apparently. So I was kinda at a loss....

Any help would be great. Im not afraid of scripting. I've got a Degree in IT (Recent Grad). But I'd prefer something with a GUI to setup. Just more comfortable with it...

I really appreciate all the answers each and everyone of you has given me though!

Flole998, 2017-12-31:
It depends on which system you want to use. You always need some kind of server, usually a raspberry pi or similar is enough, if you have a real server at home it's even better.

I can recommend using OpenHAB, it has a great Webinterface and you can control so many things with it. It integrates with Alexa and the Google home plugin is waiting for approval by google, should be available within a week.

Tuckeran5607, 2017-12-31:
I don't have access to a server unfortunately. However, I could probably run it on my parents desktop. As they keep it on like, all the time.. OpenHAB? I'll get to reading! Thanks so much.

closed 2018-02-18 by dgiese