dgiese/dustcloud #23: Choosing the firmware

freitag-13, 2017-12-31:
Hi,

thanks a lot for your work! Very well done!

How do I choose the right firmware from your repo? Does it have to be the same as already installed on the robo? Or can I choose just one of them? And what's the difference between the images? Do they just differ in different ubuntu-packages?

If I have to use the same firmware as already installed, is there a way to get the version without using the app?

Cheers!

dgiese, 2017-12-31:
Hi,
you can install any firmware you like. The exact changelogs i do not have, but i would stick with the latest version. However i am not able to upload 3134, as the firmware file is bigger than 100mbyte. Custom firmwares have also a size higher than 100Mb. As soon as i find a solution for that, i let you know ;)

linuxlurak, 2018-01-01:
@dgiese  perhaps you could upload a splitted tar.gz file?

` tar -cvvzf <archive-name>.tar.gz /path/to/folder`

`split -b 50M <archive-name>.tar.gz "parts-prefix"`

(stolen from here: https://www.linux.com/blog/split-and-join-targz-file-linux)

I'd be the happiest man on earth ;) A happy 2018 by the way to you!

linuxlurak, 2018-01-01:
ha, you are the boss! just found this:
https://dustcloud.seemoo.de/public/xiaomi.vacuum.gen1/original/encrypted/

perfect! cheers!

closed 2018-02-18 by dgiese