dgiese/dustcloud #39: Change sound files without re-flash the whole firmware

Franzl123, 2018-01-02:
Hi,
how can i change the sound files without a complete re-flash? is there an miIO.ota command?

thx

fvollmer, 2018-01-02:
According to https://github.com/marcelrv/XiaomiRobotVacuumProtocol there is a `dnld_install_sound` command. I don't know how to use it, but you should be able to use tcpdump on the robot at port 54322 to sniff the official app. I think @Flole998 has already worked out how the command works (for his app).
Another approach is to just copy them via scp.

dgiese, 2018-01-02:
Just replace the files in /opt/rockrobo/resources/sounds/prc

MeisterTR, 2018-01-02:
you can change the sound with this command:
dnld_install_sound {"md5":"#MD5#","sid":1005,"url":"http:\/\/#PFAD_DES_SERVERS#\/deutsch.pkg"}

CODeRUS, 2018-01-02:
on my vacuum changing files in /opt/rockrobo/resources/sounds/prc have no effect. should change /mnt/data/rockrobo/sounds/

linuxlurak, 2018-01-03:
you could even upload the sound files with scp:
scp /path/to/sound/files/* root@robovac-ip:/opt/rockrobo/resources/sounds/prc

dgiese, 2018-01-03:
/opt/rockrobo/resources/sounds/prc is the correct path if you did not change the language to english over the Mi Home App. Here you replace the original Chinese voice with your desired voice. The folder /mnt/data/rockrobo/sounds/ is just for downloaded sounds. As we can change the firmware, the /opt/.... folder is the better one ;)

CODeRUS, 2018-01-03:
so, if sounds exists in /mnt/data/rockrobo/sounds/ it will be used? then downloaded sounds should be wiped.

w00dst0ck, 2018-01-05:
> you can change the sound with this command:  dnld_install_sound {"md5":"#MD5#","sid":1005,"url":"http://#PFAD_DES_SERVERS#/deutsch.pkg"}

Anyone tested this with **mirobo raw**?
I would only use a german soundpack without rooting my vacuum.

CODeRUS, 2018-01-05:
> Anyone tested this with mirobo raw?

yes and it works fine

closed 2018-02-18 by dgiese

rytilahti, 2018-02-18:
Just in case someone comes looking up this topic at some later point, after https://github.com/rytilahti/python-miio/pull/153 gets merged installing a sound with python-miio works by simply calling `mirobo install_sound mysound.pkg` (assuming a fairly generic network setup and unblocked access from the vacuum to the host where the command is executed).

trananhtuan, 2019-01-31:
Sorry for digging back an old thread but how can I create sound.pkg file?

Fietspomp86, 2019-07-29:
> Sorry for digging back an old thread but how can I create sound.pkg file?

I just use the Mirobo 1.1.0 software to directly upload to the vacuum, this is quite easy to do.

trananhtuan, 2019-07-30:
That works for me too. Thank you.