dgiese/dustcloud #43: Extending internal storage

CODeRUS, 2018-01-02:
Any ideas about extending internal storage?
Can usb port be utilized for connecting pen drive?

linuxlurak, 2018-01-03:
true. true. Wanted to install some packages since apt is available. vim takes 40mb... vim-tiny only about 300kb. lucky me it's all I want at the moment (besides htop and locate). Have you inserted a stick and checked syslog? Didn't think of the usb port at all till now!

dgiese, 2018-01-03:
Hi,
so what i did for my dev firmware: I created a symlink for /var/cache (not sure about the folder name at the moment) to /mnt/data/var/... (created new folder on data partition.). I will try to script that, so you can use it. The dpkg package cache + repo information takes already 100mbyte.

USB sticks do not work at the moment, as the kernel do not have the usb storage modules.

CODeRUS, 2018-01-03:
> kernel do not have the usb storage modules
this is very sad to hear

symlink or bind mount to data partition can work for now

dgiese, 2018-01-03:
However you could compile a module for that kernel ;)
I am working on a hardware mod for storage, but it will take some time.

CODeRUS, 2018-01-03:
> you could compile a module for that kernel

info about compiling modules can be useful!

linuxlurak, 2018-01-04:
Does somebody know if the update routine (where is it? what script?) checks for modifications? I tried to implement a minimalistic http server and tested it with 0033132. After the upgrade the robot behaved very strange and in the end it talked only chinese and reverted the firmware obviously to some backup version... (no ssh etc.).  

If you are interested in the idea of a mini http server check: https://github.com/dgiese/dustcloud/commit/b207c6fc65abb0b4505f32fc899296c621fac65a

dgiese, 2018-01-04:
So i have also some strange behaviour with 3132. I tried to implement some FHEM based image for you folks, however somehow it did not work. Btw: 3132 have already the integrated english language. However i have no idea how to switch it on.

linuxlurak, 2018-01-05:
interesting. After the upgrade I had no problem in altering the files and running a http service. but I have to check again to be sure.  Hopefully there are no problems for future updates in case you have done changes to the os... Perhaps we could scan for specific binaries that check for changes. 

Yesterday I tried to check with diff if there are changes in the software.  I lack of methodology and know how to do this...

Flole998, 2018-01-06:
I use nfs for more storage. Works great.

dgiese, 2018-02-18:
see script in repo, which used data partition as storage.

closed 2018-02-18 by dgiese

Galoula, 2018-02-22:
You can use olso NFS.

Okay, only througth Wifi and its not the better way, but is possible !