dgiese/dustcloud #45: License for audio generator script and language files

fvollmer, 2018-01-03:
I just noticed we don't have any license for the audio generator script and language files (at least as far as I can see). I guess we better fix this now than later. I would suggest [GPLv3](https://www.gnu.org/licenses/gpl-3.0-standalone.html) for the script and something open like the [CC0](https://creativecommons.org/publicdomain/zero/1.0/) for the actual translations.

Obviously we need every contributor to agree:
`generate_audio.py`:
- @fquinto
- @fvollmer
- @CODeRUS 
- @rudelm

`audio_ca.csv`:
 - @fquinto 

`audio_de.csv`:
 - @fvollmer
 - @dervitti 
 - @rudelm

`audio_en.csv`:
 - @fquinto

`audio_es.csv`
 - @fquinto

`audio_pl.csv`
 - @lemmure

What are your thoughts about this?

lemmure, 2018-01-03:
Hi, accepted from my side (audio_pl.csv). 

Best regards 
Ekkehard

rudelm, 2018-01-03:
Hi,

accepted from my side for `generate_audio.py` with GPLv3 and `audio_de.csv` with CC0.

dervitti, 2018-01-03:
Hi,
GPLv3 for the script and CC0 for the actual translations is OK for me.

CODeRUS, 2018-01-03:
👍

fquinto, 2018-01-03:
Hi, accepted from my side for languages EN, ES and CA.

closed 2018-01-04 by fvollmer