dgiese/dustcloud #50: Poll: Does anyone use dustcloud implementation?

dgiese, 2018-01-04:
Just out of curiosity...

MeisterTR, 2018-01-05:
i will use the dustcloud but at this time i dont understand how to install it. in my thought. i Use a raspberry with apache where i put the www files.  the database is store all messages? where i habe to put the server.py. Next qestion. the vac communicate to the dustcloud ant the dustcloud talk to xiaomi and otherwise xiaomi to the dust and dust to to robot?

akeilox, 2018-01-05:
installation instructions would be a great help to everyone with varying degree of linux know-how, i would like to give it a try but yet to digest the big picture and inner workings. I did put up a vmware ubuntu, did apt install apache2 mysql php-myadmin but didnt go far from that the other day.

dervitti, 2018-01-05:
I´m going to...

trohnjavolta, 2018-01-06:
I definetly want to disconnect from xiaomi cloud but still be able to use tje mi home app. I didn't try dustcloud yet, never needed a database or php webserver for home use.
Could I put the whole dustcloud stuff on my rockrobo or will the database use up too much diskspace?

I wouldn't mind a simpler, easier to set up solution if there's a possibility.

dgiese, 2018-01-06:
Well, yeah. You could strip it down, that it does not record the messages anymore and use the keys from the default partition automaticly. However the main purpose of the dustcloud makes no sense to run on the vacuum. Well, i will create some example code, as it does need to have too much logic ;)

MeisterTR, 2018-01-08:
i set up a new vm with a ubuntu-server. i think everytthing was ok at installation. now i get an error while enter a new device in newdevice.php

Field 'vinda' doesn't have a default value

MeisterTR, 2018-01-08:
soo.. now the php files running... after i set all fedault value manually with phpmyadmin to NULL... i did not change the server at the robot but thats now the next step.. But should i become messages from xiaomi without changeing the server at vacuum? because at this time im not get any response from xiaomi

xwavex, 2018-01-09:
I am going to, once my robot arrives 😁

MeisterTR, 2018-01-10:
server is running and get messages from the robot but they are <32

 --------------- Thread-id: 139698152388352
192.168.178.38 via tcp wrote:
> RAW: b''
len < 32
Close Request
Class destroyed

anybody an idea?

drhirn, 2018-01-10:
I would love to. But I can't get it working right now ;)
Same problem as MeisterTR in the post above

btw.: When starting the server.py via server.sh it results in _Permission denied_. Maybe because of running as www-data?

dgiese, 2018-01-10:
Oh wait, I think www-data needs permission to run the process on port 80. Too long ago when I set the whole thing up.

drhirn, 2018-01-10:
Has it to be port 80? Another one would be more useful because when rooting, the firmware-package has to be on a webserver reachable on 80.

Kugeleis, 2018-01-10:
I want and I will use it as soon as I have time and understood how it works and how to install it. I'm not a linux pro and don't want to brick my nice new white toy.

dgiese, 2018-01-10:
Hi, yes it has to be port 80 for TCP or port 5083(or 8053...not sure at the moment) for UDP, as the vacuum will try to connect there. The UDP support of the dustcloud is a little dusty at the moment.Maybe someone can take a look at that.

dgiese, 2018-01-10:
BTW you can make a http redirection. I will post the code for that.

drhirn, 2018-01-11:
> BTW you can make a http redirection. I will post the code for that.

`iptables -t nat -A PREROUTING -p tcp --dport 80 -j REDIRECT --to-port 8080
iptables -t nat -I OUTPUT -p tcp -d 127.0.0.1 --dport 80 -j REDIRECT --to-ports 8080`

But the "len < 32"-problem seems to be the show-stopper

MeisterTR, 2018-01-11:
i just implemented a simple tcp server, not simple for me beause i not a  python writter. but the result was the same, the data contains just b''. i also seted the buffer to 1024 but there were no changes. System is a vm with a ubuntu server. So i have no more ideas now and wait till someone find a solutuion.

rudelm, 2018-01-12:
No I did not run dustcloud yet. I'm still not understanding exactly what it does? It is a proxy for the commands send to xiaomi and offers a way to block certain commands from being executed on the robot?

I would probably use it if I have a comfortable way to see the room map data and also start/stop/schedule cleanup jobs without the need of the xiaomi app.

Stevehans, 2018-01-31:
I will be as soon as my robot arrives

aebgit, 2018-02-02:
@dgiese  and colleagues thanks a lot for your efforts!!! Great Work!
...have just set it up on a dedicated VM - was not straight forward, but achievable. 
It would be really nice to have it running locally on the bot (maybe integrated into the custom-image ;-). My key-benefit is the total independence from the xiaomi-cloud.

mdallaire, 2018-02-14:
I just got it running yesterday. It was not easy to setup and required some troubleshooting, but the fact that I am now off xiaomi cloud is well worth it.

Gulaschcowboy, 2018-02-14:
Installed it right now on my RasPi, everything worked almost out of the box.

linuxlurak, 2018-02-18:
At the moment I dont use dustcloud. My robovac only communicates with [home assistant](https://github.com/home-assistant/home-assistant/releases). It is block by my firewall. 

Is there some site where I can get an overview of dustcloud's functions? Perhaps with some screens of the webui (is there any?).

fab33, 2018-02-25:
Yes some screenshot of web UI will be fine. Personnaly I'm planning to use it.

dugite-code, 2018-05-08:
I've been trying to use dustcloud but I can't figure out how to get the command server to work or how to get the upload_map.sh to work.

All I'm using it for atm is to stop my vacuum from locking up when trying to reach the xiaomi servers

closed 2018-12-05 by dgiese