dgiese/dustcloud #52: Means of communication for the dustcloud community

stefanct, 2018-01-06:
Hi,
I was waiting for something like this for some time (because proprietary vacuums either suck (no pun intended) or are very expensive as was mentioned in the 34c3 talk), thank you.
However, there is a lot of stuff to do and this issue tracker is not the best communication medium for all aspects... this I would like to propose some other means like a mailing list and/or IRC channel and making that visible in the readme. FWIW, I just joined #dustcloud on freenode's IRC but apparently I was having that idea first... :)

dgiese, 2018-01-06:
IRC? Thats soo 2000-ish like ;) How about a Telegram group?

dgiese, 2018-01-06:
https://t.me/joinchat/Fl7MmxBwXWC7ETNZAXQLSQ

Kugeleis, 2018-01-06:
+1 for telegram

stefanct, 2018-01-06:
It's your project so you choose I would say. IMHO at least reading should be possible without a dedicated client. I personally don't like telegram (requires a dedicated client and the server-side is proprietary) but the important point is to stop the issue tracker from being a web forum due to the lack of any alternatives ;)

stefanct, 2018-01-06:
Oh I just discovered that there is at least an official web client at https://web.telegram.org

mwojtynek, 2018-01-10:
Telegram invite link does not work for me. Can you create a new one?

Nebukadneza, 2018-01-17:
Hey,

I’d like to point out that telegram is a closed, centralized ecosystem that does not allow the user to choose his/her own trustworthy provider, or host your own server. This means a huge vendor-lock-in into a profit-oriented company, as well as no means of federation.

This is not what modern communication for a FOSS project should strive for! We should not exclude users, and give our communication into the hands of single points of profit-oriented entities — but rather respect the freedom of choice for the user.

Many IRC servers are run by independant, non-profit entities. On the other Hand, federated protocols like Matrix or even XMPP allow each user to communicate with every other user, while using their own trusted servers.

Best,
-Dario

dgiese, 2018-01-21:
Hi, i usually used jabber, however i think that only a fraction of people is using it. Featurewise (sending files, good working mobile client, keep tracking whats going on, etc) Telegram fits the best for the purpose. However i admit, that i would like a more open alternative, but to be honest, i did not find it yet.

Nebukadneza, 2018-01-21:
> Featurewise (sending files, good working mobile client, keep tracking whats going on, etc)
> ...
> i did not find it yet.

That is matrix! It’s kind of redeeming the mistakes made in XMPP, with overly modularifying functionality (you CAN get all of these features in xmpp, with tons of XEP support that no single entity has) — while sticking to the basic federation principle that respects the peoples freedom. You can get working server2server communication OLM (~=axolotl) end2end encryption, file/image/video/etc sending, good room management, reliable sending, etc already working. It’s a 30minute endavour to set up your own server …. All server, client and protocol are free (speech), everyone can host their own without losing the ability to communicate with everyone else!

I have taken the liberty of setting up a room at
https://matrix.to/#/#dustcloud:matrix.org
on the server hosted by the matrix-project. I’ve also linked it to irc://irc.freenode.net/#dustcloudproject …

Best,
-Dario

alext-mkrs, 2018-02-18:
I'd vote for something that doesn't require giving up your sensitive information like phone numbers, so IRC or Matrix sound better than Telegram, especially given the fact that I have no other use for Telegram. And I'd like to hang around, at least :)

dgiese, 2018-02-18:
Hey,
so i was really thinking of some solution for that. Especially i rely on a good working mobile solution (read most stuff on the phone) and a good desktop client. However i did not found a easy solution. Using Whatsapp is a nogo, Threema is costing money and does not allow sending files. IRC is somehow meh, as i did not find a nice client for desktop (windows) + phone, which does not draw tons of battery.
I wish there would be a simple solution in Form of a Telegram bot, who exports the messages to some public interface (after stripping the Usernames). If there is something like that, please contact me. Would love to find a solution for that.

alext-mkrs, 2018-02-19:
Have you had a chance to check out @nebukadneza's link? You can pick Riot web interface (first link on the landing page, iirc) to get the feel of it and then use Riot client which I guess is available on major platforms these days.

Kugeleis, 2018-02-19:
If moving to something else I'd recommend something threaded. On the telegram chat discussions cross and mess up each other. A dustcloud forum would solve this.

dgiese, 2018-12-05:
I think this issue is not relevant anymore ;)

closed 2018-12-05 by dgiese