dgiese/dustcloud #58: Control the robot with local executed scripts?

FFPeter, 2018-01-11:
Has anybody figured out how to control the (rooted) robot 'locally'? 
For example: you are logged in with ssh and can query the actual status or start the cleaning.
I would like to query the actual status and report it to my home automation if the status changes.

I know I can qery the status from remote with mirobo. But network-polling every few minutes to get data, that changes only every few days, seems not to be very smart..

commensal, 2018-05-10:
I backup the question. It there a way to send json directly to AppProxy?

vKnmnn, 2018-07-23:
if the vacuum has systemd, you could set up some systemd-timer event to post your status to the HA. I mean, It' really is just a linux box when it's rooted.
you could also write some script that translates `$: robot straight 1m` or `$: robot start clean` to the appropriate calls to 127.0.0.1 
note that i don't a roborock just yet, so this might be utter nonsense.

tuxuser, 2018-07-23:
There is this new project: https://github.com/Hypfer/Valetudo