dgiese/dustcloud #61: server.py - KeyError: 'devtype'

drhirn, 2018-01-15:
After setting up a dustcloud, I receive the following error when my robot tries to connect:

    ---------Process Thread-id: 1978647664
    ddid = 64647090, dname = saugobot, denckey = rnM5ZY6eVbQGcRJ2, full_cloud_forward = 0, forward_to_cloud = 0
    (!!!) Warning, did missmatch: 0 != 64647090
    ----------------------------------------
    Exception happened during processing of request from ('10.168.1.61', 34254)
    Traceback (most recent call last):
      File "/usr/lib/python3.5/socketserver.py", line 625, in process_request_thread
        self.finish_request(request, client_address)
      File "/usr/lib/python3.5/socketserver.py", line 354, in finish_request
        self.RequestHandlerClass(request, client_address, self)
      File "/usr/lib/python3.5/socketserver.py", line 681, in __init__
        self.handle()
      File "server.py", line 381, in handle
        self.on_read()
      File "server.py", line 456, in on_read
        process_result = self.Cloudi.process_data(self,data)
      File "server.py", line 172, in process_data
        mysocket.devtype = m.header.value["devtype"]
    KeyError: 'devtype'
    ----------------------------------------

The same happens with 'serial' if I comment out line 172 (`mysocket.devtype = m.header.value["devtype"]`).
Looks like both of them are not part of the message received from miio.protocol.

Second strange thing is the ´did missmatch´.

Nebukadneza, 2018-01-15:
Hi,

can confirm I’ve had the same behavior, when first trying to get dustcloud to run yesterday. I’d gladly assist in any debugging involved in getting this back on line.

Thanks for this great project nonetheless!
-Dario

rytilahti, 2018-01-15:
I can't say anything wrt. the did mismatch, but the devtype problem is caused by this fix: https://github.com/rytilahti/python-miio/commit/00e867d97c332e78b36f71dd156a22ebc8fb0276 . To fix it that it'd be enough to simply use `_device_id` instead of `_devtype` and `_serial`.

drhirn, 2018-01-16:
Fixed in new version of server.py.
Dustcloud now working. show.php ist showing correct info about the robot.

Thx!

closed 2018-02-01 by dgiese