dgiese/dustcloud #65: Unable to install rooted firmware to vacuum

Gronis, 2018-01-31:
I'm having trouble installing a rooted firmware on my vacuum. I have generated a custom firmware from v11_003094 downloaded from here: https://dustcloud.seemoo.de/public/xiaomi.vacuum.gen1/original/encrypted/

I'm using this soundpack: https://github.com/dgiese/dustcloud/tree/master/xiaomi.vacuum.gen1/original-soundpackages/encrypted

When installing the firmware using `./flasher.py -a <ip> -t <token> -f output/v11_003094.pkg` (but with actual ip and token), it downloads the image successfully, but fails when installing it a few minutes later with the error message: "couldn't upgrade firmware, switching to previous version".

Log from `imagebuilder.sh`

```
Running on a Mac, adjusting commands accordingly
Generate SSH Host Keys
Generating public/private rsa key pair.
Your identification has been saved in ssh_host_rsa_key.
Your public key has been saved in ssh_host_rsa_key.pub.
The key fingerprint is:
SHA256:vupJ+R1Yxf5h1vLpez/0TlrhN4j975JJUT4ablLPDJQ root@macbook.local
The key's randomart image is:
+---[RSA 2048]----+
|              .  |
|           . E  .|
|            +  o |
|           o +.o.|
|        S . + %oo|
|       o o .oB+Oo|
|      o o ..o+o**|
|     . o o .  *==|
|     .+.o .   .B@|
+----[SHA256]-----+
Generating public/private dsa key pair.
Your identification has been saved in ssh_host_dsa_key.
Your public key has been saved in ssh_host_dsa_key.pub.
The key fingerprint is:
SHA256:W4my7GpZKS9bd7HhU5r12azfeblotxS2L0xvaT2vSTk root@macbook.local
The key's randomart image is:
+---[DSA 1024]----+
|                 |
|                 |
|                 |
|         . .     |
|      ..S = o  o |
|    ..oo + O .o=o|
|     =+ o B  oE+*|
|    +o.. . . o=XO|
|   .o+.     ..*O@|
+----[SHA256]-----+
Generating public/private ecdsa key pair.
Your identification has been saved in ssh_host_ecdsa_key.
Your public key has been saved in ssh_host_ecdsa_key.pub.
The key fingerprint is:
SHA256:0R5aXROrbZM1DT/+H2osvkK0nOxsIgyLQFr5VSKe9ag root@macbook.local
The key's randomart image is:
+---[ECDSA 256]---+
|              =. |
|    . o .. . . =.|
|   o + =. + . .o+|
| .o o o .* . o.oo|
|o. . o  S + . =. |
|o  .E    *   . ..|
|. . +   +   .  ..|
| . . o . = . o. o|
|      . o oo+.  .|
+----[SHA256]-----+
Generating public/private ed25519 key pair.
Your identification has been saved in ssh_host_ed25519_key.
Your public key has been saved in ssh_host_ed25519_key.pub.
The key fingerprint is:
SHA256:m/qIZBK+6oK3QyJlWN4SInv96WyxoFfkke2oJpFQ0ig root@macbook.local
The key's randomart image is:
+--[ED25519 256]--+
| o               |
|E =              |
|oO +   o         |
|+ * o + .        |
| =.o + =S        |
|o.+.. B .o       |
|ooooo* oo        |
|o ==+.=o         |
|++o*..o..        |
+----[SHA256]-----+
decrypt soundfile
ccrypt: /Users/robin/git/dustcloud/xiaomi.vacuum.gen1/firmwarebuilder/english.pkg: key does not match -- unchanged
unpack soundfile
decrypt firmware
ccrypt: /Users/robin/git/dustcloud/xiaomi.vacuum.gen1/firmwarebuilder/v11_003094.pkg: key does not match -- unchanged
unpack firmware
fuse-ext2: version:'0.0.7', fuse_version:'27' [main (../../fuse-ext2/fuse-ext2.c:324)]
fuse-ext2: enter [do_probe (../../fuse-ext2/do_probe.c:30)]
fuse-ext2: leave [do_probe (../../fuse-ext2/do_probe.c:55)]
fuse-ext2: opts.device: /Users/robin/git/dustcloud/xiaomi.vacuum.gen1/firmwarebuilder/disk.img [main (../../fuse-ext2/fuse-ext2.c:351)]
fuse-ext2: opts.mnt_point: image [main (../../fuse-ext2/fuse-ext2.c:352)]
fuse-ext2: opts.volname: rootfs [main (../../fuse-ext2/fuse-ext2.c:353)]
fuse-ext2: opts.options: rw+ [main (../../fuse-ext2/fuse-ext2.c:354)]
fuse-ext2: parsed_options: rw,allow_other,local,noappledouble,fsname=/Users/robin/git/dustcloud/xiaomi.vacuum.gen1/firmwarebuilder/disk.img,fstypename=ext2,volname=rootfs [main (../../fuse-ext2/fuse-ext2.c:355)]
fuse-ext2: mounting read-write [main (../../fuse-ext2/fuse-ext2.c:369)]
patch ssh host keys
disable SSH firewall rule
integrate SSH authorized_keys
reconfiguring network traffic to xiaomi
pack new firmware
encrypt firmware
MD5 (output/v11_003132.pkg) = b53cbee500c817adb2a4fb4d5f6bb3cf
```
Output from `flasher.py`:

```
 ./flasher.py -a <ip> -t <token> -f output/v11_003094.pkg
Flasher for Xiaomi Vacuum
Connecting to device <ip>...
Got error when receiving: timed out
Retrying with incremented id, retries left: 3
Starting local http server...
Serving http server at <local-ip>:57104
Sending ota command with parameters: {"mode": "normal", "file_md5": "b53cbee500c817adb2a4fb4d5f6bb3cf", "install": "1", "app_url": "http://192.168.1.182:57104/output/v11_003094.pkg", "proc": "dnld install"}
Ota started!
192.168.1.83 - - [01/Feb/2018 00:42:37] "GET /output/v11_003132.pkg HTTP/1.1" 200 -
Firmware downloaded successfully.
Exiting.
```

jschmer, 2018-02-02:
Your ota command lists `/output/v11_003094.pkg` as the firmware to download but the webserver shows that `/output/v11_003132.pkg` is actually downloaded. I guess the MD5 won't match because of that mixup? Don't know why that happened though.

Gronis, 2018-02-03:
I tried both versions and I mixed them up in the provided logs. Still got the same issue using both images (with their correct md5 sums, since flasher.py takes care of that part)

andrebower, 2018-02-07:
I have exactly the same issue. Tried with 3094 and 3132. My robot is on stock 3095. Anyone has a solution?

Gronis, 2018-02-07:
I haven't tried since I opened this issue. Maybe xiaomi patched it with their latest update? Or maybe you have to disable internet while updating, to make sure the robot cannot communicate with xiaomi cloud during the update? Not sure what is causing this. Good to know that I'm not alone. My xiaomi robot is also on stock 3094.

andrebower, 2018-02-07:
Since you have to reset Wifi before the update, communication with cloud is not possible.
Another thing: I am also on MacOS and the imagebuilder.sh script does not really work so i do the steps manually, but it should still work...

Gronis, 2018-02-07:
I did not reset wifi on my vacuum. I already have the token since my vacuum is integrated in home-assistant. I got it from a android backup of xiaomi mi app. 

I'm also on MacOS. I think I had a problem where `flasher.py` was trying to find the ip address of my mac, but I just modified that function to return a hardcoded value of my ip address and that worked. I also tried the manual way but both fails

andrebower, 2018-02-07:
This message "key does not match -- unchanged" you get as the files are already decrypted?

Gronis, 2018-02-08:
I get that message sometimes and sometimes not. That might be the case. How do I know if they already are decrypted?

itelli, 2018-02-08:
Same message for me: "key does not match -- unchanged"

Probably the password "r0ckrobo#23456" is wrong / has changed?

I got the message for the english.pkg and also for the image file!

`echo "decrypt soundfile"
ccrypt -d -K "$PASSWORD_SND" "$SOUNDFILE"
mkdir sounds
cd sounds
echo "unpack soundfile"
tar -xzf "$SOUNDFILE"
cd ..
echo "decrypt firmware"
ccrypt -d -K "$PASSWORD_FW" "$FIRMWARE"
echo "unpack firmware"`

In additon, for the v2, i got the message:

`decrypt soundfile
ccrypt: /home/ts/dustcloud-master/xiaomi.vacuum.gen1/firmwarebuilder/english.pkg: key does not match -- unchanged
unpack soundfile

**gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now**`

I have unpacked the tar.gz file an manually created the sounds dic.

Any suggestions?

itelli, 2018-02-08:
Yes, they changed thas passwords!

If you use an older firmware like v11_001213.pkg (from 2017-12-13) the password **rockrobo** for the firmware still works! This versions seem to be Alpha or Beta

> Firmwares starting with v11_001xxx seem to be Alpha or Beta versions, so do not use them! They are just here for reference reasons.
https://dustcloud.seemoo.de/public/xiaomi.vacuum.gen1/original/_README_first.txt

So they also probably changed the password for the language-file. Maybe you can encrypt the already decrypted tar-file with the password and use this file :)

Gronis, 2018-02-08:
Are you able to flash the xiaomi vacuum robot with a custom firmware if these warnings does not appear? E.g you replace the files (language pack and vacuum firmware) with "new/fresh downloaded" files.

andrebower, 2018-02-08:
@itelli This problem is only related to the encryption / decryption part I guess. Did the update to the custom firmware work for you?

andrebower, 2018-02-08:
Okay I tried again with all clean and with the script and now it worked perfectly fine. I had to make one change to the script: after the fuse-ext2 disk.img image -o rw+ i added a sleep 5s because otherwise the mount was running in the background while it already tried to copy files to the not yet mounted image.

Gronis, 2018-02-09:
Which version did you try? I've tried to flash my vacuum with the original encrypted language pack (english.pkg) and firmware (v11_003094.pkg). But the update is still unsuccessful.

citruz, 2018-02-11:
I have exactly the same problem. The image building process seems to be successful, I see the firmware image being downloaded from the HTTP server but after a few minutes the robot fails with the error "couldn't upgrade firmware, switching to previous version". I tried both v11_003094 and v11_003132. I also tried a factory reset, but that does not change anything.
How did you guys figure out which version is installed on the robot? Maybe there is a new firmware version.

citruz, 2018-02-11:
I finally got it working. I used a Linux VM to create the patched firmware (instead of macOS) and that one worked.

itelli, 2018-02-15:
@Gronis Did you handle the root-process? I have a v2 and used the 001228 firmware!

Gronis, 2018-02-16:
What do you mean with `root-process`?

closed 2018-07-19 by dgiese