dgiese/dustcloud #67: TypeError: build() takes 2 positional arguments but 3 were given

knowhy, 2018-02-01:
I'm getting _TypeError: build() takes 2 positional arguments but 3 were given_ when trying to follow the instructions from the [UPDATE-howto](https://github.com/dgiese/dustcloud/blob/master/UPDATE-howto.md)

>mirobo --ip=192.168.8.1 --token=TOKEN raw_command miIO.ota '{"mode":"normal", "install":"1", "app_url":"http://192.168.8
.61/v11_003094.pkg", "file_md5":"a086f93a4cafdd3714163f940740783f", "proc":"dnld install"}'
Sending cmd miIO.ota with params {'mode': 'normal', 'install': '1', 'app_url': 'http://192.168.8.61/v11_003094.pkg', 'file_md5': 'a086f93a4cafdd37141
63f940740783f', 'proc': 'dnld install'}                                                                                                             
Traceback (most recent call last):
  File "/usr/bin/mirobo", line 11, in <module>
    load_entry_point('python-miio==0.3.4', 'console_scripts', 'mirobo')()
  File "/usr/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args[1:], **kwargs)
  File "/usr/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/miio/vacuum_cli.py", line 517, in raw_command
    click.echo(vac.raw_command(cmd, params))
  File "/usr/lib/python3.6/site-packages/miio/vacuum.py", line 270, in raw_command
    return self.send(cmd, params)
  File "/usr/lib/python3.6/site-packages/miio/device.py", line 216, in send
    m = Message.build(msg, ctx)
TypeError: build() takes 2 positional arguments but 3 were given

The _flasher.py_ script gives me the same error:

> python flasher.py -a 192.168.8.1 -t TOKEN -f v11_003094.pkg
Flasher for Xiaomi Vacuum
Connecting to device 192.168.8.1...
error while checking device: build() takes 2 positional arguments but 3 were given

rytilahti, 2018-02-01:
https://github.com/rytilahti/python-miio/pull/192 is related, the fix will be in the next python-miio release. For the time being you could downgrade construct to 2.9.22 or alternatively install python-miio from git.

knowhy, 2018-02-01:
Thanks. I downgraded construct to 2.8.22 (as 2.9.22 is not available via pip and 2.9.23 gave the same error) to Issue is resolved.

closed 2018-02-01 by knowhy