dgiese/dustcloud #68: Grab new v2-image

itelli, 2018-02-08:
Hello guys,

one question: how to grab a new v2-image? Do i have to connect the vacuum geb2 with the 001168-alpha-image first to the dustcloud and can grab a new update via dustcloud?

Many thx in advance!

MeisterTR, 2018-02-09:
now i also see the update. do you use a dustcloud? another way were the miio.log you can accept the update and turn of the wifi a few seconds later so the robot do not update and you see the message withe the firmware link

dgiese, 2018-02-09:
If you kill ccrypt before updating, then you device is not altered and you can still get the firmware. Might be useful as we dont know what funny things Rockrobo/Xiaomi did in the firmware update ;)

itelli, 2018-02-15:
Hello MeisterTR,
i want to use the dustcloud, it's already up and und running, but i couldn't add my rooted v2 device...
I have the token, the ip and anything else for the rooting (i used the 001228 from the dustcloud server), but what do i have to enter in the fields "did" and "enckey" on the newdevice.php?
Many thanks in advance!

dgiese, 2018-02-18:
both information you find after rooting in device.conf.

closed 2018-02-18 by dgiese