dgiese/dustcloud #74: server.py: TypeError: parse() takes 2 positional arguments but 3 were given

aebgit, 2018-02-13:
When running my `server.py` via `# sudo ./server.sh` I get the following error:

```
---------Process Thread-id: 140576315725568
ddid = <mydid>, dname = rockrobo, denckey = <mykey>, full_cloud_forward = 0, forward_to_cloud = 0
----------------------------------------
Exception happened during processing of request from ('<robo-ip>', 45572)
Traceback (most recent call last):
  File "/usr/lib/python3.5/socketserver.py", line 625, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.5/socketserver.py", line 354, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.5/socketserver.py", line 681, in __init__
    self.handle()
  File "/home/login/dustcloud/dustcloud/server.py", line 379, in handle
    self.on_read()
  File "/home/login/dustcloud/dustcloud/server.py", line 454, in on_read
    process_result = self.Cloudi.process_data(self,data)
  File "/home/login/dustcloud/dustcloud/server.py", line 166, in process_data
    m = Message.parse(data, ctx)
TypeError: parse() takes 2 positional arguments but 3 were given
----------------------------------------
```
According to `miio.log` the robo tries to connect to my dustcloud.

Any ideas?

Thanks in advance!

mdallaire, 2018-02-14:
I am seeing the same behavior and fixed it following this issue thread : https://github.com/dgiese/dustcloud/issues/67
(downgrade construct to version 2.8.22)

aebgit, 2018-02-15:
@mdallaire thanks a lot! I‘ll try it next weekend.

itelli, 2018-02-15:
@aebgit You can just uninstall all prev version of python-miio and install the new 0.3.6.! It's working out of the box right now!

aebgit, 2018-02-22:
@itelli TipTop, works like a charm now!

closed 2018-02-22 by aebgit