dgiese/dustcloud #83: mirobo error parsing discovery result

hs06, 2018-03-05:
Hi everyone,
after having struggled to build the image on the mac, I've succeeded and I'm at the point where I want to discover the bot. However, I fail to get the result:

```
(dustcloud-m69SZaQf) bash-3.2$ mirobo discover --handshake true
INFO:miio.device:Sending discovery to <broadcast> with timeout of 5s..
WARNING:miio.device:error while reading discover results: 'utf-8' codec can't decode byte 0xdb in position 1: invalid continuation byte
```

My system is latest High Sierra with python/pip installed via brew. LC_ALL and LANG are set to en_US.UTF-8 and exported.

```
(dustcloud-m69SZaQf) bash-3.2$ uname -a
Darwin mac 17.4.0 Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64 x86_64
(dustcloud-m69SZaQf) bash-3.2$ pip --version
pip 9.0.1 from /Users/user/.local/share/virtualenvs/dustcloud-m69SZaQf/lib/python3.6/site-packages (python 3.6)
(dustcloud-m69SZaQf) bash-3.2$ python --version
Python 3.6.4
```

Any thoughts on debugging/fixing this?

Best
Hendrik

rytilahti, 2018-03-05:
Duplicate of #77.

hs06, 2018-03-05:
Nope, no duplicate - different issue, same solution.
Fixed in python-miio 0.3.7. I had 0.3.6 (which was installed by pipenv install requests).

Manually installed from here: https://pypi.python.org/pypi/python-miio/0.3.7

Succeeded in uploading the firmware. However the vacuum told me "Failed to install firmware, installing old firmware". Any idea how to debug this? Any return code/console etc.?

dervitti, 2018-03-05:
> Succeeded in uploading the firmware. However the vacuum told me "Failed to install firmware, installing old firmware". 

please double-check that you gave the robot the right md5 that fits to the image

hs06, 2018-03-05:
Thanks for the hint. I double checked: The md5 is correct. 
```
$ cat output/v11_003132.md5 
MD5 (output/v11_003132.pkg) = 2b09a63580640d3917bfffdbd043e1e9
$ md5 output/v11_003132.pkg 
MD5 (output/v11_003132.pkg) = 2b09a63580640d3917bfffdbd043e1e9
```

and my launch-strings was
`$ mirobo --ip=192.168.8.1 --token=121739554e694d7a525749397745394a raw_command miIO.ota '{"mode":"normal", "install":"1", "app_url":"http://192.168.8.75:8000/v11_003132.pkg", "file_md5":"2b09a63580640d3917bfffdbd043e1e9","proc":"dnld install"}'`

I also tried it with the flashtool -- same story.

However, I *did* build the image in the folder 
`dustcloud/devices/xiaomi.vacuum/firmwarebuilder
`as the `xiaomi.vacuum.gen1/firmwarebuilder `was empty.

I built it on MacOSX with osxfuse. Might try to build the pkg on a Linux machine...

hs06, 2018-03-05:
Indeed, the image built under a standard Linux worked! Don’t ask me why the Mac built didn’t work. I would regard the Mac build as broken for the time being. Best! Hendrik

closed 2018-03-05 by hs06