dgiese/dustcloud #90: Broken Links in Wiki and vague hints

rudelm, 2018-03-07:
Hi,

I've just wanted to update my robot using the new automated tools which were developed over the last months. However, if I check the wiki page which points to the instructions I see only a broken link:

https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum/UPDATE-howto.md

That file isn't part of the repo anymore.

This page contains a hint to look into the subfolders:

https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto

However, that is quite vague. I've searched the repo now for any hints on the mentioned tools but I do not see them. Please add the missing file again and point users to the right place without the need to check all folders for any hints :)

Yky, 2018-03-07:
I fixed the link you mentioned here: https://github.com/dgiese/dustcloud/wiki

Also interested in the automatic tool.

rudelm, 2018-03-08:
Thank you for updating the link! I thought there would be another document which explains how the automated flashing is done.

rudelm, 2018-03-08:
I think I've found the script, I'll update the wiki page accordingly. It's the flasher.py added in [PR#47](https://github.com/dgiese/dustcloud/pull/47/files)

Yky, 2018-03-08:
I disagree. flasher.py is part of the manual update process described in the HowTo. The following sentence suggests that there is a script that is a shortcut to the manual process to do the rooting and flashing. 

"There is a automatic tool to do the rooting and flashing. Look in the subfolders."

dgiese, 2018-03-08:
uh, it might be a relict when the flasher was seperate

rudelm, 2018-03-08:
I've updated the wiki page accordingly. Its only the tool for helping you to automate the flashing. Its not for rooting. But from the wiki page I take its only about updating the robot manually with new firmware, or automatically with the flasher.py

rudelm, 2018-03-08:
Ah I see, the manual instructions do not contain anymore the instructions on how to do create a basic webserver with python. Hm so what shall we do now? I mean the flasher script is a help and automates a lot, but its not the universal one-click auto solution everyone was hoping for :)

Yky, 2018-03-08:
Regarding your first point - the HowTo - this was previously duplicated in the wiki and repository and was now removed in the repository, see issue #86

hellfi, 2018-03-08:
The scripts provided are pretty much automagic with anything Ubuntu 16.04 based.

1. get the original firmware and sound packages
2. run _imagebuilder.sh_ with corresponding arguments
3. reset vacuum's wifi, connect and run _flasher.py_ with generated output/v11xxx.pkg

rudelm, 2018-03-08:
I think this can be closed now. The sentence regarding the automated rooting is now gone from the Wiki and its only pointing to the flasher script which helps with running the necessary webserver and miio calls. The instructions on Ubuntu work and are relatively automated.

dgiese, 2018-03-08:
How about saying that we do not support Macs... that would solve a lot of problems and weird workarounds...

rudelm, 2018-03-08:
It is sad to see that its not working anymore on a Mac but I couldn't find a way to get it working again as I had the same problems as @hs06 had in https://github.com/dgiese/dustcloud/issues/83 I've already updated the instructions in hope that maybe someone else finds a solution. Otherwise it can be removed ;(

closed 2018-07-19 by dgiese