dgiese/dustcloud #97: Spelling of myCloudserverIP in server.py

kkoksvik, 2018-03-11:
Would suggest changing Line 16 to reflect correct speling of "my_cloudserver_i_p" variable, to make searching possible.

Line 16: # TODO: change myCloudserverIP to your CloudserverIP (the IP where this script is running)

Line 181: my_cloudserver_i_p = "10.0.0.1"

kkoksvik, 2018-03-11:
There is also a references to "myCloudserverIP" in README.md

-
Line 12: Example: MyCloudserverIP = "130.83.47.181"  # this is a Server of the TU Darmstadt, content does not matter

Line 37: > myCloudserverIP = "10.0.0.1"

Line 55: -4. change entries for ot.io.mi.com + ott.io.mi.com in /etc/hosts on vacuum robot to your "myCloudserverIP"

jschmer, 2018-03-12:
READMEs were moved to the wiki, updated them to be in line with the actual variable name.

PR #98 (2018-03-12): fix my_cloudserver_ip typo
Closes #97
closed 2018-03-12 by dgiese