dgiese/dustcloud #100: Error build_map

heartsaver1859, 2018-03-13:
root@raspberrypi:/home/pi/dustcloud/dustcloud# bash server.sh

Traceback (most recent call last):
  File "/dustcloud/server.py", line 33, in <module>
    from build_map import build_map
ImportError: No module named 'build_map'

I'm not able to start the server.

Yky, 2018-03-13:
You probably don't have build_map.py in /dustcloud/

You can copy /home/pi/dustcloud/dustcloud/build_map.py to /dustcloud/

Then the module should be found.

heartsaver1859, 2018-03-13:
Now i got This.... 

Traceback (most recent call last):
  File "/dustcloud/server.py", line 33, in <module>
    from build_map import build_map
  File "/dustcloud/build_map.py", line 17, in <module>
    from PIL import Image, ImageDraw, ImageChops
ImportError: No module named 'PIL'

Yky, 2018-03-13:
You need to install pillow with pip.

heartsaver1859, 2018-03-13:
Traceback (most recent call last):
  File "/dustcloud/server.py", line 33, in <module>
    from build_map import build_map
  File "/dustcloud/build_map.py", line 17, in <module>
    from PIL import Image, ImageDraw, ImageChops
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 58, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

Yky, 2018-03-13:
How did you install pillow?

heartsaver1859, 2018-03-13:
pip3 install pillow

Yky, 2018-03-13:
A quick google gives this answer: https://stackoverflow.com/questions/48012582/pillow-libopenjp2-so-7-cannot-open-shared-object-file-no-such-file-or-directo

> sudo apt-get install libopenjp2-7-dev

I didn't this problem after installing pillow, so no guarantee this helps.

heartsaver1859, 2018-03-13:
no that didn't worked out... I will give it a new try tomorrow.

Yky, 2018-03-13:
Are you using a virtualenv?

heartsaver1859, 2018-03-13:
No

Yky, 2018-03-13:
Try that then.

SteveClement, 2018-04-06:
There is something odd going on with pillow on raspi…

Also, libopenjp2-7-dev seems to not exist anymore in the stock raspian sources…

I will dig around a little, I have the same issue with another project:

```
(myPiMotion) pi@cam: ~/code/1984YourRack $ ./src/myPiMotion.py
```
```python
Traceback (most recent call last):
  File "./src/myPiMotion.py", line 8, in <module>
    from PIL import Image
  File "/home/pi/code/1984YourRack/myPiMotion/lib/python3.5/site-packages/PIL/Image.py", line 60, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
```

kbfifi, 2018-04-13:
For what it's worth: I I also had the "ImportError: No module named 'PIL'" prob. 
pip install image
fixed it for me. I'm running from Odroid C2 with DietPi