dgiese/dustcloud #99: Compile custom kernel modules

simonswine, 2018-03-13:
I'd like to make use of custom kernel modules. Has anyone managed to cross compile working modules.
That's my try so far, but I am not able to load the modules

https://github.com/simonswine/rockrobo-kernel/commits/master

simonswine, 2018-03-13:
Ok it finally worked for me. Now able to mount a USB pen drive:

```
[    8.215472] usbcore: registered new interface driver usb-storage
[    8.215487] USB Mass Storage support registered.
[    9.211051] scsi 0:8:0:0: Direct-Access     Intenso  Ultra Line       1.00 PQ: 0 ANSI: 6
[    9.213372] sd 0:8:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.4 GiB)
[    9.214040] sd 0:8:0:0: [sda] Write Protect is off
[    9.214065] sd 0:8:0:0: [sda] Mode Sense: 03 00 00 00
[    9.214408] sd 0:8:0:0: [sda] No Caching mode page present
[    9.221656] sd 0:8:0:0: [sda] Assuming drive cache: write through
[    9.231022] sd 0:8:0:0: [sda] No Caching mode page present
[    9.237073] sd 0:8:0:0: [sda] Assuming drive cache: write through
[    9.244696]  sda: sda1
[    9.246628] sd 0:8:0:0: [sda] No Caching mode page present
[    9.252785] sd 0:8:0:0: [sda] Assuming drive cache: write through
[    9.259497] sd 0:8:0:0: [sda] Attached SCSI removable disk
[    9.747403] EXT4-fs (sda1): recovery complete
[    9.750109] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
```

```
root@rockrobo:/media/usb0# df -h
Filesystem       Size  Used Avail Use% Mounted on
/dev/root        494M  369M  100M  79% /
devtmpfs         202M  4.0K  202M   1% /dev
none             4.0K     0  4.0K   0% /sys/fs/cgroup
tmpfs             30M  4.0K   30M   1% /tmp
none              50M  240K   50M   1% /run
none             5.0M     0  5.0M   0% /run/lock
tmpfs            100M  2.3M   98M   3% /run/shm
none             100M     0  100M   0% /run/user
/dev/mmcblk0p6    16M  1.2M   14M   8% /mnt/default
/dev/mmcblk0p11   16M  1.3M   14M   9% /mnt/reserve
/dev/mmcblk0p1   1.6G  323M  1.2G  22% /mnt/data
/dev/mmcblk0p10  494M  221M  248M  48% /mnt/updbuf
/dev/sda1         15G  2.8G   11G  20% /media/usb0
```

If you want to try the binary modules you can find them here:

https://github.com/simonswine/rockrobo-kernel/releases/tag/v11_003096

Let me know how it goes

simonswine, 2018-03-13:
Speed could be quicker:

```
root@rockrobo:/media/usb0# dd if=/dev/zero of=/media/usb0/testfile bs=1M count=1024 oflag=direct
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB) copied, 73.8829 s, 14.5 MB/s
```

EliasKotlyar, 2019-03-16:
I have tried to compile some drivers for USB-Cameras (UVC), and wanted just to share that its currently not possible - see my comment here : https://github.com/simonswine/rockrobo-kernel/issues/1