dgiese/dustcloud #103: Other ways to get a shell except wifi?

devkid, 2018-03-26:
Hi,

I'd love to play around a little bit more with the firmware, maybe extracting the robot services into a Docker containers etc. so that I could easily update the unterlying Ubuntu to a more recent version.

But I'm afraid that somehow I'm going to brick the device and will have no way to login anymore. Has anyone found another way to get a root shell on the device except for the unlocked SSH login over Wifi? I own the Roborock S50 and have seen in your photos that there is also a USB port hidden under some covers. Can that one be used to get a shell somehow? (Maybe after enabling it via some bootloader flags?) If not, are there functional UART pins on the mainboard that are accessible without completely tearing apart the device?

Galoula, 2018-03-26:
Hi,

Yes !

On he's firmware you have one with ADB modified.

I have put his "adbd" file into mine version, and I can get root shell througth USB port (adb shell).

devkid, 2018-03-26:
Ok, so this will give me a shell in case I break the wifi somehow. Are there other ways to get a shell in case I flash an unbootable firmware image or an image that fails to start adbd? Or talking to the bootloader directly to flash something?

I read in the presentation slides that the device has two partitions and flashes one after the other in the update process. (The second partition only if it successfully boots the first one? How does it detect "successful"? Can I switch between the partitions?)

dgiese, 2018-03-28:
I *think* recovery should work always. So if you put the adbd binary into recovery, your should always be able to go back... at least as long as you do not mess around with the STM32 and its firmware.

When i remember correctly, triggering the watchdog sets a uboot variable to boot the passive copy of the system. If the watchdog is triggered again, the recovery will be restored. However, this requires the watchdog to work, and as it is a piece of software in the firmware, it may not work always.

devkid, 2018-03-28:
Is there any documentation about the recovery? How would I force the device into recovery mode? Buttons? Or by sending something on the USB port while it's booting?

dgiese, 2018-03-31:
I will test the corresponding buttons later. If i remember correctly, its the following:
Gen1: Hold Home Button and press WiFi-reset button
Gen2: Hold Spot Cleaning and Home Button, and press WiFi-reset button
(instead of WiFi reset, you might also just power on the vacuum)

lordmochy, 2018-04-29:
"Gen2: Hold Spot Cleaning and Home Button, and press WiFi-reset button
(instead of WiFi reset, you might also just power on the vacuum, e.g. press Power button 5 seconds to switch completely off, dock the vacuum) "

It dosen't work for me, can u explain it better? I just installed a root firmware and now I cant access to my vacuum. The light of the wifi is always off :(

peoro, 2018-06-19:
Could you guys possibly confirm that it's possible to factory-reset the Xiaomi Roborock S50? Did anybody successfully try that?

@lordmochy did you have any luck? From what I read you should hold the main button, and press the reset one (the small close to the WiFi LED, you need a pen/toothpick to use it).

I'd like to root my robot, but before proceeding I'd love to have a first hand confirmation that I can easily bring the thing back to life even if I managed to screw up everything and make the system unbootable.

dgiese, 2018-06-19:
yes, works.
see here: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset

lordmochy, 2018-06-19:
I just can`t. Probably my roborock was broken. I had a new one now

dgiese, 2018-06-19:
Uh how so? Never heard of the case, where the robot was not recoverable (if it was docked, buttons holded and reset pressed)

dgiese, 2018-07-19:
did not hear any update, close for now

closed 2018-07-19 by dgiese