dgiese/dustcloud #106: Possible attack on the ciphering key

X-Ryl669, 2018-04-03:
If I understand correctly, once provisioned, the device sends some packet containing JSON ciphered with a **static** *device-specific* AES key.
What ciphering suite is being used ? EBC ? CTR ?

Because if it's any of those 2, then it can be possible to figure out the key by monitoring the WIFI's network traffic.
Technically, the ciphered packet's bytes are `pkt(i) = msg(i) XOR transformedKey(i)` and since msg(i) is '{' for `i == 0`, and '}' for `i == msg.length-1`, then you can figure out the `transformedKey(i)` byte-by-byte while sending packets of different size (for example, by changing the SSID length on your WIFI router)

So the step by step guide would be:

1. Prepare some wifi AP with TCP dump capability (or use a WIFI sniffer)
2. Unprovision your vacuum
3. Set up the WIFI's on your vacuum while capturing all packets
4. Extract the `transformedKey(0)` and `transformedKey(msg.length -1)` from the equation above
5. Change the SSID's name to one more byte
6. Set up wifi again
7. Extract `transformedKey(msg.length - 1)` for the next byte of the AES key
8. Repeat until you've the complete AES key.

Obviously, this fails if the AES key changes (which, I don't think it does as I understand it), or if the IV used is based on some random event.

Can someone confirm the ciphering scheme used ?

nuschpl, 2018-04-23:
What about WiFi encryption or why would one attack it's own device this way having direct hardware access? Provisining and unprovisioning is very time consuming

closed 2018-07-19 by dgiese

nuschpl, 2018-07-20:
Why did you close this?