dgiese/dustcloud #109: English voice for goto and zone cleaning in v1 with 003194 firmware

ovaltineo, 2018-04-06:
I uploaded the latest English pkg with the 003194 firmware to my v1 robot.  Everything works fine, except that the voices for goto and zone cleaning are still in Chinese.  Assuming that this works on the v2 robot (I haven't seen any complaints about it), then I think that the voice ids for these functions on v1 is different from the v2.

akeilox, 2018-04-06:
Yes they are missing in v1 stndrd english pack, I had to use the Update on the English voice pack (in mi app) after which missing/chinese voice prompts updated to English for the zone cleaning.

dugite-code, 2018-05-20:
Looks like it would be simple to add this to the [audio_en.csv](https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum/audio_generator/language/audio_en.csv)

Only five zone related files not in that csv
```
root@rockrobo:~# ls /opt/rockrobo/resources/sounds/en | grep zone
resume_zone.wav
stop_zone.wav
zone.wav
zone_complete.wav
zone_failed.wav
```