dgiese/dustcloud #112: Dustcloud at boot

Galoula, 2018-04-15:
Hi, I arrive to get dustcloud directly inside the robot (totaly autonomous without Internet).

But I search a way to start dustcloud at boot.
I have put in my /etc/rc.local:

iptables -t nat -I OUTPUT --dest 1.2.3.4 -p udp --dport 8053 -j DNAT --to-dest 10.33.20.11:8053
iptables -t nat -I OUTPUT --dest 1.2.3.4 -p tcp --dport 80 -j DNAT --to-dest 10.33.20.11:80
chroot /mnt/data/DustCloud/ /etc/init.d/mysql start
chroot /mnt/data/DustCloud/ /etc/init.d/apache2 start
su - root -c "screen -dm chroot /mnt/data/DustCloud/ /dustcloud/server.sh"

All works, but the last line doesn't works well:
It's tarts, but nothing works. If i kill the screen and restart it in SSH, it's works...

Other way to start "chroot /mnt/data/DustCloud/ /dustcloud/server.sh" without rc.local ?

I think the is environment variable missing to get it works fine...

Galoula, 2018-04-15:
Hi, I have find a workaround:
I have added in /opt/rockrobo/watchdog/ProcessList.conf this line:
dustcloud.sh,setsid dustcloud.sh >> /mnt/data/rockrobo/rrlog/miio.log 2>&1&,0,1,0

I have created a file /opt/rockrobo/miio/dustcloud.sh with this content:
#!/bin/bash
chroot /mnt/data/DustCloud/ /dustcloud/server.sh

I have also created a /root/KillDustCloudAtBoot file with:
#!/bin/bash
killDustCloud()
{
	date > /tmp/kill.date
	sleep 30
	kill $(pgrep python3)
	date >> /tmp/kill.date
}
killDustCloud &

And add in my rc.local:
/root/KillDustCloudAtBoot

Reboot the robot, and can get my DustCloud works at boot time :-)

closed 2018-04-15 by Galoula

dugite-code, 2018-05-04:
@Galoula I attempted to install python3 and pip on my vacuum but ran out of space using apt. How did you set this up?

did you `chroot /mnt/data` prior to any apt installs or did you do the install manually

dugite-code, 2018-05-07:
Ok so my main issue was I had installed nano doing an `apt-get clean` cleared some root space but still apt installs are space risky.

I found:
```
apt-get --download-only -o Dir::Cache="/mnt/data/dustcloud" \
    -o Dir::Cache::archives="./" install python3

dpkg -i /mnt/data/dustcloud/*.deb
```
unfortunately this is python 3.4 not python 3.5 as python-miio calls for. I tried the [deadsnakes repo](https://launchpad.net/%7Edeadsnakes/+archive/ubuntu/ppa) lacks the armhf builds so I'm stuck atm.