dgiese/dustcloud #113: RC command in web interface

kbfifi, 2018-04-16:
Hi thank you for this wunderfull project! I've got everything working. I'm trying to figure out how to get the app_rc_move to work. Tried several things:
Method: app_rc_move
Params: [{"omega": -0.8 , "velocity": 0.29 , "seqnum": 1 , "duration": 5000 }]

I guess the question is how to format commands with parameters.
Any hints? Quotes? block brackets?

bravehurts2, 2018-04-20:
I did not try to control by dustcloud but did you try it with the app_rc_start command first?

Json to vac to start the session and to move. 
{"method":"app_rc_start" params:[]}
{"method":"app_rc_move" params:[{"omega":-0.8,"velocity":0.29,"seqnum":1,"duration":5000}]}

Dont trust the remote control. Your vac will end in different positions for the same command sequences.

bravehurts2, 2018-04-23:
I use my own solution and send the crypted package directly to the vac. The json I posted is the payload string I send before crypting.