dgiese/dustcloud #118: Steps after installing dustcloud and rooting device?

OhSoGood, 2018-04-29:
Hi,
I've read and reread the wiki, but I'm not sure of the necessary steps to perform after having installed (and configured) dustcloud and rooted the vacuum.

What I did (see previous post), was simply to change the wifi settings of the vacuum, expecting it to connect to dustcloud. Is that ok? What else should I do?

Also, can the officiel mi app connect to dustcloud? If so, how?

Thanks for your help, I'm a little lost...

OhSoGood, 2018-05-05:
sniff...nobody?

dugite-code, 2018-05-08:
My experience:
* I found I could only get my vacuum talking to dustcloud using UDP
* I haven't gotten dustcloud to talk to the command server, so the buttons don't work.
* zone cleaning is currently not implemented on python--miio https://github.com/rytilahti/python-miio/pull/178 and it requires a newer firmware then the on in the repo and I haven't found a download for the latest firmware as I set up my vacuum without any Xiaomi app interaction.
* I never got the Flole app because there was no way to manually enter the token.
* I'm still trying to figure out how to get the map off the Vacuum automatically.

I'm looking into using [homeassistant](https://www.home-assistant.io/) for controlling everything as it looks like it's got the most development for ease of use

dugite-code, 2018-05-08:
I Just noticed the firmware is actually 003194 so ignore that comment

skamoen, 2018-05-08:
I went through the setup process yesterday, including setting up dustcloud but I don't think it's actually required if you never use the Xiaomi app. Directly after turning it on I rooted it following these steps: https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto and got the new token with `printf $(cat /mnt/data/miio/device.token) | xxd -p`

In my firewall I blocked all traffic except local traffic, so using `python-miio` or [Home-Assistant](https://www.home-assistant.io/components/vacuum.xiaomi_miio/) I can control it remotely. 

I'm stil not sure where dustcloud comes into play, I assume that's if you set it up through the cloud to begin with.

dkm, 2018-05-08:
@skamoen How did you change the vaccum network configuration ? After first start, its has its own open wifi network (unprovisioned state). IIUC, you can configure it to connect to your own local network (using miio command), but it still expects to connect to Xiaomi cloud. If not, then miio command are not accepted.

I would really like to avoid having to setup dustcloud or connecting to xiaomi cloud, and of course, I don't want to leave it unprovisioned. I have HomeAssistant setup-up and ready, but I still have to figure out how to connect vacuum on my network and still accept commands...

jschmer, 2018-05-08:
@skamoen I think there were reports that the vacuum won't accept any remote commands anymore (e.g. with python-miio) if it doesn't have a connection to the cloud servers for some time. That's where dustcloud comes into play with simulating a cloud server for the vacuum.
The main purpose in my opinion is to peek into the messages sent between cloud and vacuum though, so I see the dustcloud server more like a (reverse-)engineering/debugging tool. It just happens to have some benefits for normal use too.

skamoen, 2018-05-08:
@dkm I configured my network through the miio command after setting up a static DHCP lease so I could pre-emptively block everything in the firewall. After that, it was unresponsive for a few hours, like @jschmer described. 

I'm not sure what changed, but a reboot might have fixed that. It's not a 100% reliable, and I'm really not sure what triggered it. I know I did a manual reboot, but I can't recall if it worked right after that. Anyway, it accepts commands now and I'm very sure it did not have cloud connectivity after rooting it. I did have Dustcloud running when it started responding again, but Dustcloud is reporting it never got any messages so I don't think that was it.

dkm, 2018-05-09:
@skamoen I'll try to setup the network and then reboot the vacuum. I'll keep track of everything so that we can have a better understanding of what really works. How far has this issue been studied ? The pdf is a bit vague, it nearly looks like it has not been really checked. I'll happily try to be more thorough as I guess being able to setup the vacuum without internet connectivity with the minimal number of steps will be useful :D

rytilahti, 2018-05-09:
For connectivity issues, maybe one of those mentioned in https://github.com/home-assistant/home-assistant/issues/11048#issuecomment-375930041 is relevant? If one dropping everything on the router, the vacuum tries to keep connecting and waits for some timeouts, that's why either 1) rejecting instead of dropping and/or 2) letting DNS through, seems to help in this case.

That being said, it is not completely bulletproof, and one can see some timeouts happening when looking into the miio log on the device. I'm personally running mine on my separate network with no remarkable problems even when the robot has no internet access. I'm just allowing DNS through and rejecting with `icmp-port-unreachable` otherwise.

dkm, 2018-05-10:
So, I've tried to follow these steps, but now that the vacuum is connected to my wlan, I can't send it any command :D I'll wait a bit to see if some timeout kicks in... I tried to keep track of everything in : 

https://gist.github.com/dkm/21de05a620bb325959c619af8f542e61

TL;DR; robot is connected to wlan, it can use DNS but doesn't have internet access. Not possible to use miio commands.

dkm, 2018-05-10:
I'm not sure how to check that my token is correct... I'm not sure how to verify it using /mnt/data/miio/device.token

dugite-code, 2018-05-10:
'''
printf $(cat /mnt/data/miio/device.token) | xxd -p
'''

Regards

Dugite-Code

Art, Tech and Information. What's the diff?

https://peekread.info

PGP Fingerprint: f978 c9ce a65a 3d32 dc28 d32c 5170 dbcd c547 f257
PGP Public Key: https://gist.github.com/dugite-code/83737e5fcd077659dc171c2eec50d17d

dugite-code, 2018-05-10:
```
printf $(cat /mnt/data/miio/device.token) | xxd -p
```

Just be aware you may need to reboot after the initial wifi connection for the token to be correct

dkm, 2018-05-10:
@dugite-code Ok, that was the problem... The token I had was not correct anymore. Using token extracted as shown above just works. At least I can send `status` command and the vacuum replies, so I'm confident. Maybe I can reformat/rephrase my steps above and put them in the wiki.

skamoen, 2018-05-10:
So we confirmed the token changes when you connect to your wifi. I'm sure I used the new token for my commands when it wasn't responding, so it's still possible a reboot is necessary after connecting to the wifi. Not sure how we can test this without going through the whole setup process again.

oleid, 2018-05-21:
Does anyone know, how to make the robot talk to dustcloud? I installed their docker images, registered the robot using the data from `/mnt/default/device.conf` at SERVER_IP:81, however it never connects to that server. This is my hosts file on the robot:

<pre>
# cat /etc/hosts
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

0.0.0.0       awsbj0-files.fds.api.xiaomi.com
0.0.0.0       awsbj0.fds.api.xiaomi.com

192.168.0.250	ot.io.mi.com ott.io.mi.com
</pre>

So I assumed, that the robot will connect to 192.168.0.250, yet, dustcloud's web interface claims the last contact was `0000-00-00 00:00:00`. So what am I missing?

JackGruber, 2018-05-21:
Your host entries should point to a public IP address like 130.83.47.181

# On the robot
**/etc/hosts**
```
130.83.47.181       ot.io.mi.com
130.83.47.181       ott.io.mi.com
130.83.47.181       sg.ot.io.mi.com
130.83.47.181       sg.ott.io.mi.com
```

than add a iptables entry for redirect the traffic to your dustcloud IP
```
iptables -t nat -A OUTPUT -p tcp --dport 80 -d 130.83.47.181 -j DNAT --to-destination 192.168.0.250:80
iptables -t nat -A OUTPUT -p udp --dport 8053 -d 130.83.47.181 -j DNAT --to-destination 192.168.0.250:8053
```

oleid, 2018-05-21:
Ah, I would have thought that a hosts entry pointing to the correct IP would be enough. Thanks, that did it for me. :)

Radio-van, 2018-07-09:
Hello.
I faced a trouble with connecting vacuum to dustcloud. Here is what I've done:
- I flashed my vacuum (1st gen) with custom firmware (no troubles with building and flashing) with `--disable-xiaomi` option. Flashing was succesfull, I'm able to ssh vacuum.
- I used python-miio to configure wifi, but after that vacuum fast blinking blue led and mirobo commands with new token (from `/mnt/data/miio/device.token`) don't work saying that discover failed (that's because vacuum can't connect to Xiaomi cloud, right?).
- I run three docker containers with db, phpMyAdmin and Dustcloud. I'm not very familiar with databases, but I imported one from the dustcloud github repo, created a user, grant access and it seems to work. I added vacuum to database, but dustcloud says that the last contact was 0000-00-00 00:00:00.
- Then I found this issue and edited `/etc/hosts` on vacuum, now it looks like this:

```
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

0.0.0.0       awsbj0-files.fds.api.xiaomi.com
0.0.0.0       awsbj0.fds.api.xiaomi.com

130.83.47.181       ot.io.mi.com
130.83.47.181       ott.io.mi.com
130.83.47.181       sg.ot.io.mi.com
130.83.47.181       sg.ott.io.mi.com
```
Btw, am I right, that 130.83.47.181 doesn't matter, it could be any public IP?
And then I added 2 rules to my OpenWrt router (it has internet access):
```
iptables -t nat -A OUTPUT -p tcp --dport 80 -d 130.83.47.181 -j DNAT --to-destination 10.0.2.1:80
iptables -t nat -A OUTPUT -p udp --dport 8053 -d 130.83.47.181 -j DNAT --to-destination 10.0.2.1:8053
```
`10.0.2.1` is address of dustcloud instance.
Then I've been waiting for 17 hours, but vacuum still didn't connect to dustcloud.
Previously I flashed it with same custom firmware, but without `--disable-xiaomi` option and configured with MiHome app and vacuum worked well, so I think I missed something when I tried to connect it to dustcloud.

JackGruber, 2018-07-12:
Can you show us your docker run command for the dustcloud container.

Radio-van, 2018-07-12:
Sure
```
docker run --name dustcloud -d --link dustcloud_mariadb:db \
-p 80-81:80-81/tcp -p 8053:8053/udp -p 1121:1121/tcp \
-e CMDSERVER=10.0.2.1 \        #IP of RPi with dustcloud container
-v /tmp/data:/dustcloud/data \
jackgruber/dustcloud
```
The only differences with run command from documentation are `CMDSERVER` ip and no `TZ `env, because Resin.io doesn't have `/etc/timezone`.

JackGruber, 2018-07-12:
for the Timezone option you can youse every TZ from https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

what the docker log show? 
Command `docker logs dustcloud `

Radio-van, 2018-07-16:
Well, after starting dustcloud, there is nothing in logs.

Here is full log from dustcloud container:
```
Start v1.3.3
===================
CLOUDSERVERIP=130.83.47.181
MYSQLSERVER=db
MYSQLDB=dustcloud
MYSQLUSER=dustcloud
MYSQLPW=*****
CMDSERVER_PORT=1121
CMDSERVER=10.0.2.1
COUNTRYSERVER=ott.io.mi.com
TZ=Europe/Berlin
===================
Clearing any old processes...
Starting apache...
VirtualHost configuration:
AH00558: httpd: Could not reliably determine the server's fully qualified domain name, using 172.17.0.6. Set the 'ServerName' directive globally to suppress this message
ServerRoot: "/var/www"
Main DocumentRoot: "/var/www/localhost/htdocs"
Main ErrorLog: "/var/www/logs/error.log"
Mutex default: dir="/run/apache2/" mechanism=default 
Mutex mpm-accept: using_defaults
PidFile: "/run/apache2/httpd.pid"
Define: DUMP_VHOSTS
Define: DUMP_RUN_CFG
User: name="apache" id=100
Group: name="apache" id=101
AH00558: httpd: Could not reliably determine the server's fully qualified domain name, using 172.17.0.6. Set the 'ServerName' directive globally to suppress this message
Starting Dustcloud...
```
Actually, I don't need dustcloud, because I control vacuum with `mirobo`+`home assistant`, but I wished vacuum to be isolated from Xiaomi servers. So, I needed dustcloud only for initial setup (because without connection to cloud vacuum don't listen to mirobo commands) and I solved problem by commenting
```
0.0.0.0       awsbj0-files.fds.api.xiaomi.com
0.0.0.0       awsbj0.fds.api.xiaomi.com
```
lines and letting vacuum to connect to Xiaomi cloud, and then revent the lines back. Now vacuum works well with `mirobo` and seems to not contacting Xiaomi.
But I really like the idea of dustcloud and think tha it would be great to solve connection problem.

dkm, 2018-07-16:
FWIW, my roborock has never been connected to the internet (filtered on router) and I've never had any issue to set it up with Home Assistant. So I'm not sure about the :
> So, I needed dustcloud only for initial setup (because without connection to cloud vacuum don't listen to mirobo commands)

Radio-van, 2018-07-16:
Hm.. Well, in my case, vacuum wasn't discoverable by mirobo (even with correct token) until I let it connect to cloud once.
First I didn't know why it doesn't working, but then I read @jschmer post
 > I think there were reports that the vacuum won't accept any remote commands anymore (e.g. with python-miio) if it doesn't have a connection to the cloud servers for some time. That's where dustcloud comes into play with simulating a cloud server for the vacuum.

and tried to connect vaccum to dustcloud first (unsuccessful) and then to Xiaomi cloud (successful). After initial connection vacuum receives mirobo commands well.

JackGruber, 2018-07-16:
i haven't tried to use the iptables rules on a router yet, only on the my vacuum.
Cloud you try to setup the rules on the vacuum?

eifinger, 2018-09-26:
I just found out that my vacuum would not respond after connecting to my wifi as long as my PiHole blocked the xiaomi servers. The vacuum does not have access to the internet but needs to resolve the dns names as it seems.
```
ot.io.mi.com
ott.io.mi.com
```

igrcic, 2018-11-04:
Hello,

i had similar issues (using docker containers myself)
After some trial and error, i figured out these two hosts also have to be resolvable:

130.83.47.181       awsbj0-files.fds.api.xiaomi.com
130.83.47.181       awsbj0.fds.api.xiaomi.com

Now im getting my vacuum connected to dustcloud server. 

Tnx again for this  fun project guys :)

emilzaman, 2019-01-07:
I delete by mistake the /mnt/default/device.conf And after that, I restarted the device. Now I can't connect to robot anymore and the open wifi is named "NOCONFIG_MIAPFFEC". I've tried to reset to factory defaults but no luck.
Do I have any chance to recover the file or to connect to it again via ssh, somehow?
Tnx