dgiese/dustcloud #119: starting server.sh, asking for password for www-data

WeareGermans, 2018-05-03:
Hi,

i installed dustclud using this howto on stretch https://github.com/dgiese/dustcloud/wiki/Dustcloud-Raspberry-Pi-Installation .

I came to the last step using sudo as user pi for almost all steps.
How do you start the server.sh?
If i start it i get asked for a password for the user www-data, which is not possible to enter due to user restrictions.

Thanks

OhSoGood, 2018-05-05:
The script server.sh is only running server.py as user www-data. In a first time at least, you can directly execute the latter with:
`python3 /dustcloud/server.py`

WeareGermans, 2018-05-05:
Thanks, that worked.
I think i even got the original problem solved by adding a sudo up front.

closed 2018-05-05 by WeareGermans