dgiese/dustcloud #130: WLAN problems

JackGruber, 2018-05-25:
My robot keeps losing the WLAN connection and connects again after a while. Anybody else have that problem?

```
May 21 16:38:24 rockrobo kernel: [45841.836927] RTL871X: cfg80211_rtw_scan(wlan0)
May 21 16:38:24 rockrobo kernel: [45841.837349] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 0 ms
May 21 16:38:24 rockrobo kernel: [45841.837532] RTL871X: SetHwReg8188E(wlan0): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x48c=0x00000000
May 21 16:38:25 rockrobo kernel: [45843.196742] RTL871X: SetHwReg8188E(wlan0): [HW_VAR_MACID_WAKEUP] macid=0, org reg_0x48c=0x00000001
May 21 16:38:25 rockrobo kernel: [45843.198108] RTL871X: survey done event(10) band:0 for wlan0
May 21 16:38:25 rockrobo kernel: [45843.198236] RTL871X: rtw_indicate_scan_done(wlan0)
May 21 16:38:47 rockrobo kernel: [45864.900866] RTL871X: ERROR rtw_check_bcn_info: start new beacon (seq=2334)
May 21 16:38:47 rockrobo kernel: [45864.900893] RTL871X: ERROR rtw_check_bcn_info: cur beacon key
May 21 16:38:47 rockrobo kernel: [45864.900910] RTL871X: rtw_dump_bcn_keys: ssid =
May 21 16:38:47 rockrobo kernel: [45864.900926] RTL871X: rtw_dump_bcn_keys: channel = 6
May 21 16:38:47 rockrobo kernel: [45864.900942] RTL871X: rtw_dump_bcn_keys: ht_cap = 19ef
May 21 16:38:47 rockrobo kernel: [45864.900958] RTL871X: rtw_dump_bcn_keys: ht_info_infos_0_sco = 3
May 21 16:38:47 rockrobo kernel: [45864.900979] RTL871X: rtw_dump_bcn_keys: sec=3, group = 10, pair = 10, 8021X = 0
May 21 16:38:47 rockrobo kernel: [45864.900997] RTL871X: ERROR rtw_check_bcn_info: new beacon key
May 21 16:38:47 rockrobo kernel: [45864.901014] RTL871X: rtw_dump_bcn_keys: ssid =
May 21 16:38:47 rockrobo kernel: [45864.901030] RTL871X: rtw_dump_bcn_keys: channel = 6
May 21 16:38:47 rockrobo kernel: [45864.901045] RTL871X: rtw_dump_bcn_keys: ht_cap = 19ad
May 21 16:38:47 rockrobo kernel: [45864.901061] RTL871X: rtw_dump_bcn_keys: ht_info_infos_0_sco = 0
May 21 16:38:47 rockrobo kernel: [45864.901081] RTL871X: rtw_dump_bcn_keys: sec=3, group = 10, pair = 10, 8021X = 0
May 21 16:38:47 rockrobo kernel: [45865.003323] RTL871X: ERROR rtw_check_bcn_info: new beacon again (seq=2338)
May 21 16:38:47 rockrobo kernel: [45865.105669] RTL871X: ERROR rtw_check_bcn_info: new beacon again (seq=2341)
May 21 16:38:47 rockrobo kernel: [45865.105693] RTL871X: ERROR rtw_check_bcn_info: new beacon occur!!
May 21 16:38:47 rockrobo kernel: [45865.105711] RTL871X: rtw_check_bcn_info bw mode change
May 21 16:38:47 rockrobo kernel: [45865.105730] RTL871X: rtw_check_bcn_info bcn now: ht_cap_info:19ef ht_info_infos_0:7
May 21 16:38:47 rockrobo kernel: [45865.105752] RTL871X: rtw_check_bcn_info bcn link: ht_cap_info:19ad ht_info_infos_0:4
May 21 16:38:47 rockrobo kernel: [45865.105843] RTL871X: UpdateHalRAMask8188E => mac_id:0, rate_id:0, networkType:0x0b, mask:0x000fffff
May 21 16:38:47 rockrobo kernel: [45865.105858]          ==> rssi_level:0, rate_bitmap:0x000ff015
May 21 16:38:54 rockrobo kernel: [45871.927125] RTL871X: cfg80211_rtw_scan(wlan0)
May 21 16:38:54 rockrobo kernel: [45871.927830] RTL871X: SetHwReg8188E:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 0 ms
May 21 16:38:54 rockrobo kernel: [45871.928044] RTL871X: SetHwReg8188E(wlan0): [HW_VAR_MACID_SLEEP] macid=0, org reg_0x48c=0x00000000
May 21 16:38:55 rockrobo kernel: [45872.697415] RTL871X: Drop duplicate management frame with seq_num = 2581.
May 21 16:38:55 rockrobo kernel: [45873.282530] RTL871X: SetHwReg8188E(wlan0): [HW_VAR_MACID_WAKEUP] macid=0, org reg_0x48c=0x00000001
May 21 16:38:55 rockrobo kernel: [45873.283945] RTL871X: survey done event(13) band:0 for wlan0
May 21 16:38:55 rockrobo kernel: [45873.284114] RTL871X: rtw_indicate_scan_done(wlan0)
M
```

Muyz, 2018-09-22:
Yes, I have. Just received my S50 today, rooted it and find the WiFi link to be unstable. Found these messages as well running dmesg. Have you solved this?

JackGruber, 2018-09-23:
sorry, no

oblogic7, 2019-02-23:
Just rooted my S50 and have found that it drops the wifi connection as well.  Has anybody figured out why this is happening?

cryptomilk, 2019-02-23:
This has been fixed by Roborock in firmware v001720, however the robot drives around like it is drunk with that firmware ...

oblogic7, 2019-02-23:
Interesting.  I flashed with v001792.  Seems that it drops the wifi connection after it has been docked for a certain amount of time.  I've only noticed because the robot stops responding to my commands coming from Home Assistant.  When this happens, my wifi controller shows that the device is no longer connected.  I'll do some more investigating to see if I can identify why this is happening.

oblogic7, 2019-02-23:
I started a cleaning after my last comment and started experiencing the drunken driving that you mentioned.  Guess I should downgrade to < 1720 and deal with wifi connectivity issues only.

[image]

dgiese, 2019-03-01:
I think the 1702 has some kind of bug. Had experienced the same problems, but they have been gone after 1720. However I have no idea about the map, as I only reverse the stuff... ;)

closed 2019-04-28 by dgiese