dgiese/dustcloud #131: new firmware 3.3.9_001412 for vaccum Gen2

Petititi, 2018-05-26:
Hi,
there is a new firmware for the Gen2 roborock. How can I help you to root this firmware?

Flole998, 2018-06-07:
Try to root and see what happens.

Petititi, 2018-06-16:
It's working like a charm! My only concern now is that the port 80 redirection (or 8053) doesn't seems to work anymore. Nothing is redirected to my dustcloud server. I'll try to dump my network traffic to see what's going on.

Petititi, 2018-06-16:
My vaccum seems to call an https server on amazon:
`ec2-35-156-208-231.eu-central-1.compute.amazonaws.com.https`

Do you think the redirection using iptable could still works?

Flole998, 2018-06-16:
That is the europe server if I am not mistaken, the redirection instructions are for china mainland so it does not work when using a different server.

Petititi, 2018-06-16:
@Flole998 you're right, it's eu-central... Anyone have some experiencies in man-in-the-middle for https connexion? Or a good tutorial I could follow? I think if we change certificats on the vaccum, we would be able to do this (but dustcloud server would then have to be a bit more complex, isn't it?).

Flole998, 2018-06-16:
That is not necessery , there should be no need to mess with all that. The connection to the command server is definitely not https, not entirely sure what you caught there, probably the upload of something.

Flole998, 2018-06-16:
@marcelrv  We need to seperate here between phone and App, there is a huge difference! The vacuum does not use HTTPS for commands, just for uploading data to china.

dgiese, 2018-07-19:
Version is rootable, closing issue

closed 2018-07-19 by dgiese