dgiese/dustcloud #140: How to get latest firmware .pkg from Xiaomi which is not listed here?

ovaltineo, 2018-09-03:
My Gen 1 Xiaomi is telling me that there is 3.3.9_003357 available firmware.  But this firmware is not listed in the .pkg files in this project.  How can I pull the latest .pkg from Xiaomi so I can update my rooted vacuum?

lukeman11, 2018-09-06:
Yeah, same for my Roborock. Floleapp is telling me, there is 3.3.9_001598. The last one in this project is 3.3.9_001518 :-(

ovaltineo, 2018-09-06:
I found out how to do it.  First, disable update on the robot by renaming /usr/bin/ccrypt to /usr/bin/ccrypt_.  Then, use the app to update the firmware.  It will download the firmware but will fail to update.  The .pkg file can now be copied from /mnt/data/.temp/Update.pkg.

muhviehstah, 2018-09-07:
v1 firmwares can be found here https://cloud.it-kraut.net/s/BW4SHMeTRt36k43

muhviehstah, 2018-09-21:
latest v1 firmware 3358 added

dugite-code, 2018-09-25:
It would be great if anyone who has the latest firmware could submit the encrypted/decrypted hashes for updating the firmwareversions.xlsx

@muhviehstah files have these MD5 hashes
35538ab7035b4061efed0b10a90e8f69 *v11_003290.pkg
e03ed1f5bb5cf34bffda7b8e6202e4a1 *v11_003357.pkg
32707314f898c8bd0e631f98013fca06 *v11_003358.pkg

ovaltineo, 2018-10-10:
Gen 1 v11_003504 from Flolevac is here https://drive.google.com/uc?id=1vwvMu44NZ8W_JiqULyf2S91lIo2lUVHQ&export=download

junior013, 2018-10-26:
If you know the version number, you need (from FloleApp or elsewhere), you can download it from official direct link, like this: https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_001228.pkg

dugite-code, 2018-10-30:
@junior013 I'm having no luck fetching any other firmware using the suggested link with any of the latest package names

https://cdn.awsbj0.fds.api.xiaomi.com/rubys/updpkg/v11_003290.pkg
https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_003357.pkg
https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_003358.pkg

All return:
`Object Not Found: Make sure your object exist in current region, request-id=#######################`

Given how old the package number in your link is it could be no longer used or it could be the cdn for my region just isn't allowing me to see any of the files. Where did you find this link?

junior013, 2018-10-30:
Sorry, it really doesn't work with your versions :/ I used it for latest Gen2 FW-s from EU, and downloaded 4 different version. (Maybe it's only for G2?)
I found the link in a thread on roboter-forum.de

dugite-code, 2018-10-30:
It could well be "rubys" is the model name for the Gen2? A search online suggests Xaiomi uses `[url]/[model]/[resource]` so it might be possible.

StefanSchoof, 2018-11-15:
Has someone tried the 3.3.9_001633 for the gen2? The description mentions saved maps and software barrier. That sound interesting.

FeatureExpert, 2018-11-15:
Seems those features are disabled on EU servers unfortunately.

reaper7, 2018-11-16:
@FeatureExpert - it means that root works on gen2 with fw 001633

reaper7, 2018-11-16:
@junior013 - tnx for the info :)

StefanSchoof, 2018-11-16:
I updated to 001633 and for me the saved map option was in the app. I had not to edit the lab.cfg, but the already contains a 1

reaper7, 2018-11-16:
@asch8505 - in Poland I have 5.4.29

blakeblackshear, 2018-11-27:
Found the path to download the firmware for Gen1 directly: https://cdn.awsbj0.fds.api.mi-img.com/updpkg/v11_003416.pkg

dugite-code, 2018-11-27:
@blakeblackshear Huh well that's interesting they must have set that up prior to their current layout of `[url]/[model]/[resource]` they appear to be using for their other products. So I didn't think of just removing `[model]`

Thanks for letting us all know

dugite-code, 2018-11-27:
[I've added a page to the wiki](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Firmware-download-from-Xiaomi-Servers)

dgiese, 2018-12-05:
I changed the site, and added more information: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Firmware

closed 2018-12-05 by dgiese

sanghpm, 2018-12-12:
hi bro , thank good i finded you . 
i have a problem with my xiaomi gen 1 , a problem at EMMC it empty. im finding the mentos to conect xiaomi vancuum to pc but dont have any thing guide about that . 
i has download the firmware v11_003357.pkg but i cant update.
everyone helpt me about that .
thank you so much

dgiese, 2018-12-12:
Not related to this issue.
However eMMC problems might have 2 general causes:  1. The eMMC is completely broken (e.g. not detected or detected with 0 MB). In that case the only way would be to replace the eMMC (like with a SD card). 2. The partition format or data on the eMMC is deleted. I would not think that this is the reason for your case, but theoretically you could restore the data from a working gen1 robot.

In both cases your vacuum will propably never be able to connect to the cloud again, as the DID, Key and MAC information is lost. In addition the LDS calibration data is also stored on the eMMC, this can also cause issues. Maybe it would be possible to run the vacuum with dummycloud and Valetudo, as it is independant of the cloud. 

The first step to figure out what happened to the eMMC would be to connect to serial and look into the bootlog.

sanghpm, 2018-12-25:
Yeah thank Dgiese somuch about the answer me .
I have the vancuum xiaomi and that is life , i had read emmc by UFI and write a new emmc after that . but appear in my eye the vancuum not really life because dont turn on wifi it only turn on and clearner and came back home . 
I had read more topic about update the firmware by dumycloud but really i dont have any solution to connect with my vancuum
Can you give me some solution about update firmware to vancuum , ADB or Write by UFI

dgiese, 2018-12-25:
I guess you need to connect via Serial to figure out whats going on.

sanghpm, 2018-12-26:
Hey , can you add friend me on facebook . we can tell in there . ( https://www.facebook.com/sanghpm )

pepeEL, 2020-06-24:
How i can update firmware to my Xiomi S5 gen 2 from file *.pkg  ?