dgiese/dustcloud #141: How to put Vacuum Robot Gen1 in unprovisioned mode

tlpbu, 2018-09-12:
I have built firmware for my Vacuum Robot Gen1 and ...
[Upload manual](https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto#upload-the-firmware-image) says "6. Now press the wifi button on your vacuum" but there is not Wifi button on device only reset pinhole.

Can anyone clarify how to get Vacuum Robot Gen1 in unprovisioned mode to be able to upload firmware?

merlink01, 2018-09-12:
You must reset the robot. See this Thread:

https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset

merlink01, 2018-09-12:
Oh yes, I think I did it like this:

Hold down the home button and then with a pencil hold down the reset button (while still holding the home button down). It may take a couple tries.

tlpbu, 2018-09-13:
Thanks @merlink01! That worked.

closed 2018-12-05 by dgiese