dgiese/dustcloud #142: Xiaomi vacuum cleaner cliff sensors

Venedikt1988, 2018-09-14:
Does anyone know how to programming switch off vacuum cleaner cliff sensors? One sensors seems to broke up and it constantly showing error to clean the cliff sensors. Cannot fix the problem in service as they don't have them to replace.

UomoNero2099, 2019-10-26:
Did you find a solution? I have the same problem

Venedikt1988, 2019-10-27:
Yes, i dismantled it. As long as i do not need cliff sensors i just openned the casing of all cliff sensors and took out the immitting and recieving part and stick them together with scotch. Already working more than one year. The problem seems that one of the infra red diod lost his immision and lit very weak so that reciever cannot see it.

Venedikt1988, 2019-10-27:
Each sensor separetly with its immitting part. I did not try to find the one eefective, so i did it with all sensors. You can easily find wich one is a problem by cctv camera in ir mode

closed 2019-10-27 by Venedikt1988