dgiese/dustcloud #144: Add new soundpackages

tl-pierre, 2018-09-18:
Hello,
First, thank you for this incredible project.
Since few weeks there are new official soundpackages for xiaomi vacuums (Italian/French/Spanish/German), would it be possible to add them in the folder `devices/xiaomi.vacuum/original-soundpackages/` like the english soundpackage ?

Thanks in advance

miguelangel-nubla, 2018-09-24:
@tl-pierre can you share a reference to those official sound packages?

tl-pierre, 2018-09-25:
Hi,
On the [Mi Home application](https://play.google.com/store/apps/details?id=com.xiaomi.smarthome) (and surely on [FloleVac](https://play.google.com/store/apps/details?id=de.flole.xiaomi)) it is possible to download them (video example [here](https://www.youtube.com/watch?v=68v9ao3fhp0)). 
I wonder if it's possible to download the files from a vacuum still connected to the xiaomi cloud with [the command](https://github.com/marcelrv/XiaomiRobotVacuumProtocol/blob/master/install_sound.md)
`dnld_install_sound {"md5":"#MD5#","sid":1005,"url":"http://PATH TO SERVER/FILE.pkg"}`
What do you think ?

miguelangel-nubla, 2018-09-25:
I am unable to reproduce the list in the video.
Mine shows chinese, engligh, and other chinese voices only.
MiHome 5.4.14 mainland china server, vaccum gen 2 firmware 3.3.9_1518
Also tried using eu server with no luck.

Do you know if that list is for gen 1 only or something like that??

tl-pierre, 2018-09-25:
I don't know, sorry :/ Can it be related to [that](https://github.com/dgiese/dustcloud/tree/master/devices/xiaomi.vacuum.gen2/CN_to_EU_script) ?
If anyone can help us with these soundpackages problems it would be super nice :)

miguelangel-nubla, 2018-09-25:
Don't think so.
Mine came EU from factory and list was the same.

Does your list show those packages? What versions are you using?

tl-pierre, 2018-09-25:
I don't use Mi Home, my roborock S50 is rooted and not connected to internet (that's why I need the soundpackages in ".pkg" format).

FeatureExpert, 2018-10-17:
I guess you have to be connected to EU servers to be able to see and install sound packages other than english and chineese. You can always download it manually and install it using, for example, miboro.

closed 2018-12-05 by dgiese