dgiese/dustcloud #147: Latest firmware?

olskar, 2018-09-29:
I see that the latest firmware is on this github  for V1 is v11_003254.pkg

however the latest version is v11_003416.pkg

Could anyone upload that so I can root it and upload it to my vacuum? :)

(Or can I extract it somehow and upload it myself?)

dugite-code, 2018-10-01:
See: https://github.com/dgiese/dustcloud/issues/140#issuecomment-424211278 @dgiese has stated there are currently not well tested

dugite-code, 2018-10-30:
https://github.com/dgiese/dustcloud/issues/140#issuecomment-433404820

dugite-code, 2018-11-29:
How to download directly from Xiaomi is now documented [in the wiki](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Firmware-download-from-Xiaomi-Servers)

olealgoritme, 2018-12-04:
v11_003422.pkg was the latest I could find, when I looked yesterday.

https://cdn.awsbj0.fds.api.mi-img.com/updpkg/v11_003422.pkg

olskar, 2018-12-05:
> v11_003422.pkg was the latest I could find, when I looked yesterday.
> 
> https://cdn.awsbj0.fds.api.mi-img.com/updpkg/v11_003422.pkg

https://cdn.awsbj0.fds.api.mi-img.com/updpkg/v11_003452.pkg today :)

FeatureExpert, 2018-12-05:
But be careful, highest number != latest version

dgiese, 2018-12-05:
Please see here: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Firmware
I would close this issue for now.

closed 2018-12-05 by dgiese