dgiese/dustcloud #151: Send firmware OtA: BrokenPipeError: [Errno 32] Broken pipe

philippegabert, 2018-10-26:
Hi everyone, 

I am trying to follow [this guide](https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto) to root my vacuum (Gen 2)

I've create the package and I am trying to upload it to the vaccum but I got this message:

```
(.venv) philippe@philippe-virtual-machine:~/dc$ python dustcloud/devices/xiaomi.vacuum/firmwarebuilder/flasher.py -f output/v11_001518.pkg -t 5565xxxxxbxxxxxxxxxxxxxxxxxxxxxx (Token masked here)
Flasher for Xiaomi Vacuum
Address is not set. Trying to discover.
Connecting to device 192.168.8.1...
Starting local http server...
Serving http server at 192.168.8.157:39811
Sending ota command with parameters: {"mode": "normal", "install": "1", "app_url": "http://192.168.8.157:39811/output/v11_001518.pkg", "file_md5": "dacc365a2155f997cc2206d8d5895f12", "proc": "dnld install"}
Ota started!
192.168.8.1 - - [26/Oct/2018 00:12:34] "GET /output/v11_001518.pkg HTTP/1.1" 200 -

Exception happened during processing of request from ('192.168.8.1', 57715)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 639, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.6/socketserver.py", line 696, in __init__
    self.handle()
  File "/usr/lib/python3.6/http/server.py", line 418, in handle
    self.handle_one_request()
  File "/usr/lib/python3.6/http/server.py", line 406, in handle_one_request
    method()
  File "/usr/lib/python3.6/http/server.py", line 639, in do_GET
    self.copyfile(f, self.wfile)
  File "/usr/lib/python3.6/http/server.py", line 800, in copyfile
    shutil.copyfileobj(source, outputfile)
  File "/usr/lib/python3.6/shutil.py", line 82, in copyfileobj
    fdst.write(buf)
  File "/usr/lib/python3.6/socketserver.py", line 775, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe
Got error when receiving: timed out
Traceback (most recent call last):
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 257, in send
    data, addr = s.recvfrom(1024)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 257, in send
    data, addr = s.recvfrom(1024)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 257, in send
    data, addr = s.recvfrom(1024)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 257, in send
    data, addr = s.recvfrom(1024)
socket.timeout: timed out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "dustcloud/devices/xiaomi.vacuum/firmwarebuilder/flasher.py", line 206, in <module>
    main()
  File "dustcloud/devices/xiaomi.vacuum/firmwarebuilder/flasher.py", line 194, in main
    ota_progress = vacuum.send('miIO.get_ota_progress')[0]
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 286, in send
    return self.send(command, parameters, retry_count - 1)
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 286, in send
    return self.send(command, parameters, retry_count - 1)
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 286, in send
    return self.send(command, parameters, retry_count - 1)
  File "/home/philippe/dc/.venv/lib/python3.6/site-packages/miio/device.py", line 289, in send
    raise DeviceException("No response from the device") from ex
miio.exceptions.DeviceException: No response from the device
```

So... it seems to start something and to be able to communicate with the vaccum... but the part to get the status of the upload is failing because of the connection being terminated...

Also, I don't know if it could lead to something interesting, but I've seen that when I run the discover method with the handshake flag, I receive a different token each time...

Any idea on how to solve this?

Thanks !

Philippe

herrwusel, 2018-11-25:
Can confirm this with my rockrobo s50

herrwusel, 2018-11-25:
@philippegabert check your firmware on your vacuum. I did it and I got it working by chosing the according firmware from here: https://dustcloud.seemoo.de/public/temp/ 
Rebuilding the firmware made the upload not fail.

dugite-code, 2018-11-30:
@philippegabert you can [download directly from Xiaomi](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Firmware-download-from-Xiaomi-Servers) if you know the firmware number. For 3.3.9_001633 the link is: https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_001633.pkg