dgiese/dustcloud #152: Map data relocated

junior013, 2018-10-26:
On my Gen2 with latest FW (3.3.9_001644) the map extraction don't working anymore. The SLAM file still updating in /run/shm directory, but navmap*.ppm is empty (exists, updating, but 0 byte size).
In MiHome, the map still working, so it stored somewhere else.
Can somebody, with more knowledge correct this issue?

reaper7, 2018-11-16:
@junior013  - you know new map location?

junior013, 2018-11-16:
@reaper7 No, its more complicated. The navmap files in /run/shm are still made, but empty (0 size). There are some map files in somewhere /mnt/data (sorry, i can't access my robot now), like lastmap, usermap, etc., but the file format differs too, so the old converter can't read them.
Somebody with better knowledge needed

reaper7, 2018-11-17:
what is stored in robot.db?
there is a interesting table 'cleanmaps' with timestamp when the vacuum cleaner was working
and contains blobs with some gz data which can be easily unpacked:
e.g.
```
sqlite3 robot.db "select hex(map) from cleanmaps where begin=1542465567" | xxd -r -p > robotout.gz

or

sqlite3 robot.db "select writefile('robotout.gz', map) from cleanmaps where begin=1542465567"                                       
```
inside gz is a one file which looks like a map but with unknown for me file header (hex 0x72 0x72 0x14) 
and some more complicated data at the file end

junior013, 2018-11-17:
@reaper7 Its known. That is the cleaning history file. It stores all the finished cleanaps. The data is the map bitmap and all the path data after.
But, the actual running, "live" map is not in that db :(

reaper7, 2018-11-17:
I did not know about it

FeatureExpert, 2018-11-17:
@junior013 Roborock sends live map files to xiaomi cloud servers i.e. when you use the MiHome app. You can capture these files but it's a bit complicated and requires some knowledge.

junior013, 2018-11-18:
@FeatureExpert, I know, I used the map extractor of dustcloud - till it worked.
I don't have enough knowledge, so I hope the original developer correct this feature sometime...

Sennevds, 2020-02-05:
Hi,

With the latest gen2 firmware (2008) the map data extractor doesn't work.
I've searched to fix this and found out that there's some changes to the binary file:
https://github.com/rand256/valetudo-mapper/blob/f9cb790d79d13ab835b437293d693ea5d863211b/lib/RRMapParser.js#L41
I've tried his solution and could extract the image from the robot.db with this

junior013, 2020-02-05:
@Sennevds Sorry, the topic is more than a year old. After a while, I also started to use Valetudo and it works without problems.
Now, I close it. After all, thank you for the idea!

closed 2020-02-05 by junior013