dgiese/dustcloud #156: unable to download FW

leonardpitzu, 2018-11-30:
https://dustcloud.seemoo.de keeps asking for username:password

leonardpitzu, 2018-11-30:
i figured the answer (RTFM...).

closed 2018-11-30 by leonardpitzu

rytilahti, 2018-11-30:
Coud you be a bit more specific? Which FM? Some other people in the future may encounter this issue, so it's good idea to document it.

leonardpitzu, 2018-12-01:
I got a new toy, namely the roborock 2, aka s50. I wanted to root it (by the way, many thanks for your work otherwise probably i would have never achieved my goal) and followed the wiki. At one point, in the rooting section, the reader is pointed to dustcloud.seemoo.de in order to download the firmware. dustcloud.seemoo.de asks for a username and a passoword so downloading using this method is not possible. Further down in the wiki a new download method is mentioned - one that downloads directly from xiaomi's servers. One thing missing though... Let's say a have FW version 3.3.9-001633 (this version i read from the original app). Now i want to download the same so i can root the vacuum. For that i need to download some file named v11_00xxxx. I found no info (or maybe i missed it) but xxxx stands for the last 4 digits of the firmware you want to download, so in my example it's v11_001633.pkg. One little thing that might be useful is a list/table of sorts with the rootable (aka verified) versions + which is the latest version released by xiaomi.

dugite-code, 2018-12-03:
@leonardpitzu the firmware 3.3.9-001633 link would be: https://cdn.awsbj0.fds.api.mi-img.com/rubys/updpkg/v11_001633.pkg You just replace the version number i.e. `3.3.9-` with `v11_`

As for tested firmware there is the [firmwareversions.xlsx](https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum.gen1/firmwareversions.xlsx) for the gen 1 with MD5 Hashes but not for gen 2

I added some table for tested firmware in the wiki if you want to add your experiences [Tested Firmware](https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Firmware-download-from-Xiaomi-Servers#tested-firmware)

dgiese, 2018-12-03:
The storage server is down at the moment. Thats the reason why the firmware is not downloadable at the moment. Hopefully it is repaired soon. In the meantime use the urls here: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Firmware-download-from-Xiaomi-Servers

The firmwareversion.xlsx is very outdated. After there were new versions every week, I stopped to update it. Propably I should remove it...