dgiese/dustcloud #158: 3.39_001633 v2 firmware

greenimpala, 2018-12-12:
New firmware is out - has anybody had success rooting with this?

FeatureExpert, 2018-12-12:
3.3.9_001633 is "old" firmware, you can get root without any issues

akeilox, 2018-12-13:
Can’t find clarification if following can be still rooted:

V2 firmware v 3.3.9-001702
and
V1 firmware v ...3542

greenimpala, 2018-12-13:
> 3.3.9_001633 is "old" firmware, you can get root without any issues

Old as in 4 weeks ago or am I misunderstanding - as far as I know this was released recently (see [post](http://en.miui.com/thread-4531308-1-1.html) on official forum).

dgiese, 2018-12-25:
Check the temp folder link on the wiki page about Firmware. The filedates reflect the real date when the firmware was build.

nicoh88, 2018-12-30:
#175

closed 2019-04-09 by dgiese