dgiese/dustcloud #164: Unable to SSH via putty to Roborock V2

blueacidic, 2018-12-19:
Was able to initially and followed info on this link: [https://xiaomirobot.wordpress.com/xiaomi-robot-roborock-mainland-china-continental-europe/](url)

However, because I translated from french to english the formatting change where I had to run these commands: 

I ran the following:

mount / mnt / default / -o remount, rw
rm -Rf / mnt / default / sounds
rm -Rf / mnt / default / roborock.conf
exit

instead of:

mount / mnt/default/ -o remount,rw
rm -Rf /mnt/default/sounds
rm -Rf /mnt/default/roborock.conf
exit

And didn't realize till I updated the roborock to the latest firmware and now I cant seem to SSH. Any tips would be appreciated.

blueacidic, 2018-12-19:
Currently running on 3.3.9_001633 firmware

cryptomilk, 2018-12-19:
If the robot updates itself, then the root is gone. Normally updates should be block to prevent the robot from installing updates.

blueacidic, 2018-12-20:
Meaning I should only leave it on the rooted firmware and not run any updates?

dgiese, 2018-12-25:
Yep, you need to patch every update you doing, if you want to maintain root.

closed 2018-12-25 by dgiese