dgiese/dustcloud #169: Problem mit Anleitung

AlexanderSch90, 2018-12-26:
Hallo Dennis, 
ich komme auf dein Angebot zurück und beschreibe dir mein Problem hier einmal ausführlich.

Ich möchte die Firmware analog deiner Anleitung bauen. Nutze dafür eine Ubuntu 18.04 virtuelle Maschine, welche seitens der Virtualisierungsumgebung keine mir bekannten Einschränkungen hat. Zudem bin ich als Root angemeldet. Trotzdem kann ich das Image anscheinend nicht richtig mounten. 

Zwei Infos noch zu deiner Anleitung: der Punkt 3. `
Download the dustcloud firmwarebuilder: wget https://github.com/dgiese/dustcloud/releases/download/0.1/firmwarebuilder_0.1.zip
Unzip the file unzip firmwarebuilder_0.1.zip
` führt dazu, dass die imagebuilder.sh nicht im Unix-Format abgespeichert wird. Zudem besitzen die Dateien nicht die passenden Rechte (0644 statt 0755). Aber auch nachdem die Rechte gesetzt sind funktioniert es leider nicht. Anbei der ganze Vorgang (die Rechte und die Firmware habe ich per WinSCP durchgeführt, deshalb kein Log...)

```
login as: root
Using keyboard-interactive authentication.
Password:
Welcome to Ubuntu 18.04.1 LTS (GNU/Linux 4.15.18-9-pve x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Canonical Livepatch is available for installation.
   - Reduce system reboots and improve kernel security. Activate at:
     https://ubuntu.com/livepatch

0 Software-Pakete können aktualisiert werden.
0 Aktualisierungen sind Sicherheitsaktualisierungen.

You have mail.
Last login: Tue Dec 25 20:47:32 2018 from 10.4.4.60
root@Staubsauger:~# apt-get install ccrypt git wget unzip
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
unzip ist schon die neueste Version (6.0-21ubuntu1).
ccrypt ist schon die neueste Version (1.10-6).
git ist schon die neueste Version (1:2.17.1-1ubuntu0.4).
wget ist schon die neueste Version (1.19.4-1ubuntu2.1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@Staubsauger:~# cd dc
root@Staubsauger:~/dc# cp ~/.ssh/id_rsa.pub .
root@Staubsauger:~/dc# ls -la
insgesamt 111140
drwxr-xr-x  3 root root      4096 Dez 26 08:25 .
drwx------ 20 root root      4096 Dez 26 08:21 ..
-rw-r--r--  1 root root     84396 Feb 28  2018 adbd
-rw-r--r--  1 root root   6563318 Dez 25 20:52 english.pkg
-rw-r--r--  1 root root      7414 Aug 15 07:18 flasher.py
-rw-r--r--  1 root root       730 Dez 26 08:25 id_rsa.pub
-rw-r--r--  1 root root     12368 Dez 26 08:24 imagebuilder.sh
-rw-r--r--  1 root root     35821 Feb 28  2018 LICENSE.md
drwxr-xr-x  2 root root      4096 Dez 26 08:23 unprovisioned
-rw-r--r--  1 root root 107067695 Dez 25 20:49 v11_003254.pkg
root@Staubsauger:~/dc# ls -la
insgesamt 111140
drwxr-xr-x  3 root root      4096 Dez 26 08:25 .
drwx------ 20 root root      4096 Dez 26 08:21 ..
-rwxr-xr-x  1 root root     84396 Feb 28  2018 adbd
-rwxr-xr-x  1 root root   6563318 Dez 25 20:52 english.pkg
-rwxr-xr-x  1 root root      7414 Aug 15 07:18 flasher.py
-rwxr-xr-x  1 root root       730 Dez 26 08:25 id_rsa.pub
-rwxr-xr-x  1 root root     12368 Dez 26 08:24 imagebuilder.sh
-rwxr-xr-x  1 root root     35821 Feb 28  2018 LICENSE.md
drwxr-xr-x  2 root root      4096 Dez 26 08:23 unprovisioned
-rwxr-xr-x  1 root root 107067695 Dez 25 20:49 v11_003254.pkg
root@Staubsauger:~/dc# ./imagebuilder.sh -f v11_003254.pkg -s english.pkg -k id_rsa.pub -t Europe/Berlin --disable-xiaomi
Scriptpath: .
/root/dc/imagebuilder.sh
compatible readlink found!
Generate SSH Host Keys
Generating public/private rsa key pair.
Your identification has been saved in ssh_host_rsa_key.
Your public key has been saved in ssh_host_rsa_key.pub.
The key fingerprint is:
SHA256:2HpF4MspNNG1XjqrKBwgM9E76HK9xDGg2zcr1Xncmbs root@Staubsauger
The key's randomart image is:
+---[RSA 2048]----+
| .    .....      |
|. o    o.. .     |
| + o  o . o .    |
|* + o. = = o     |
|.* = +ooS.=o     |
|o.o O ooo.+o     |
|.. = =... ..     |
|  . =  o ..      |
|   . .. . E.     |
+----[SHA256]-----+
Generating public/private dsa key pair.
Your identification has been saved in ssh_host_dsa_key.
Your public key has been saved in ssh_host_dsa_key.pub.
The key fingerprint is:
SHA256:jhYsJX9IcFVcsseizPrzbDGQsuSzxJyPG1/dWTz+f7g root@Staubsauger
The key's randomart image is:
+---[DSA 1024]----+
|    . ...oo..    |
|     o    .+     |
|    . o  .o o    |
|     *o+o. o   . |
|    .==+S.     .o|
|     .O*  o. ..o.|
|     .==. .o. oo |
|     .o+oo.   . o|
|      ..o+o   E.=|
+----[SHA256]-----+
Generating public/private ecdsa key pair.
Your identification has been saved in ssh_host_ecdsa_key.
Your public key has been saved in ssh_host_ecdsa_key.pub.
The key fingerprint is:
SHA256:V+GHZxbWx7HC385uIaG/4GuJ32c/3e72FQgWk6kcafo root@Staubsauger
The key's randomart image is:
+---[ECDSA 256]---+
|          .o+ o+.|
|         + +++ .=|
|        + oo+o=..|
|       . o...*+ .|
|        S . ...o.|
|         E . . +.|
|          ..o . B|
|         ..oo. =*|
|          o+.o+=X|
+----[SHA256]-----+
Generating public/private ed25519 key pair.
Your identification has been saved in ssh_host_ed25519_key.
Your public key has been saved in ssh_host_ed25519_key.pub.
The key fingerprint is:
SHA256:ZD/MmaeZ+5yYJo5WcuWwTlxfNw/L31H57n8Eh47UgPc root@Staubsauger
The key's randomart image is:
+--[ED25519 256]--+
|           .     |
|          . o    |
|        o  . + ..|
|       o.+oo. Eo+|
|       .S*Bo.= *=|
|      . * .*o +.+|
|       *  +    +o|
|      ..o .= .  =|
|     ....o+.+  .=|
+----[SHA256]-----+
decrypt soundfile
ccrypt: /root/dc/english.pkg: key does not match -- unchanged
unpack soundfile
decrypt firmware
ccrypt: /root/dc/v11_003254.pkg: key does not match -- unchanged
unpack firmware
mount: image: das Einhängen ist fehlgeschlagen: Vorgang nicht zulässig.
patch ssh host keys
./imagebuilder.sh: Zeile 236: ./etc/ssh/ssh_host_rsa_key: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 237: ./etc/ssh/ssh_host_rsa_key.pub: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 238: ./etc/ssh/ssh_host_dsa_key: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 239: ./etc/ssh/ssh_host_dsa_key.pub: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 240: ./etc/ssh/ssh_host_ecdsa_key: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 241: ./etc/ssh/ssh_host_ecdsa_key.pub: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 242: ./etc/ssh/ssh_host_ed25519_key: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 243: ./etc/ssh/ssh_host_ed25519_key.pub: Datei oder Verzeichnis nicht gefunden
disable SSH firewall rule
sed: ./opt/rockrobo/watchdog/rrwatchdoge.conf kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
integrate SSH authorized_keys
mkdir: das Verzeichnis »./root/.ssh“ kann nicht angelegt werden: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf './root/.ssh' nicht möglich: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 256: ./root/.ssh/authorized_keys: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf './root/.ssh/authorized_keys' nicht möglich: Datei oder Verzeichnis nicht gefunden
reconfiguring network traffic to xiaomi
./imagebuilder.sh: Zeile 264: ./etc/hosts: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 265: ./etc/hosts: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 376: ./opt/rockrobo/watchdog/ntpserver.conf: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 377: ./opt/rockrobo/watchdog/ntpserver.conf: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 378: ./opt/rockrobo/watchdog/ntpserver.conf: Datei oder Verzeichnis nicht gefunden
./imagebuilder.sh: Zeile 379: ./etc/timezone: Datei oder Verzeichnis nicht gefunden
cp: das angegebene Ziel './opt/rockrobo/resources/sounds/prc/' ist kein Verzeichnis
umount: image: nicht eingehängt.
waiting for unmount...
umount: image: nicht eingehängt.
waiting for unmount...
^C
root@Staubsauger:~/dc#
```
PS: Die VM heißt Staubsauger ;-)

Besten Dank

flinke-flasche, 2018-12-28:
Hast du dir mal die Anleitung von haus-Automatisierung.com angeschaut?
Ging bei mir ohne Probleme mit einem ubuntu 18. 
https://youtu.be/t3b2TN9nlKY
 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

AlexanderSch90, 2018-12-28:
@flinke-flasche Ich habe genau die Anleitung befolgt. Das Problem ist jedoch wohl ein anderes. Ich hatte es versucht in einem LXC-Container auszuführen und nicht in einer "echten" VM. Jetzt geht es...

Danke euch!

closed 2018-12-28 by AlexanderSch90

Philipp97714, 2019-04-25:
Ich habe das selbe Problem. Gibt es inzwischen eine Lösung um nicht eine VM nutzen zu müssen?

cryptomilk, 2019-04-25:
Entweder das loop Kernel modul ist nicht geladen oder loop mount geht im container nicht.

marciogranzotto, 2019-08-03:
I'm having the same problem on a Mac

marciogranzotto, 2019-08-03:
Appears to be a problem with kext slots on macOS. Unloading some kext fixed the problem

Reference: 
https://github.com/osxfuse/osxfuse/issues/315#issuecomment-258456598