dgiese/dustcloud #170: Portuguese language (Portuguese, European (pt-PT))

jamlazevedo, 2018-12-27:
[audio_pt.zip](https://github.com/dgiese/dustcloud/files/2711208/audio_pt.zip)

fabiomsouto, 2019-01-29:
This was helpful. I've generated the audio using OSX's `say`. Also fixed a couple of typos
I'm sharing the small script I've done in PHP to do this.

```php
<?php

const FILENAME = 'audio_pt.csv';

if (($handle = fopen(FILENAME, "r")) !== FALSE) {
    while (($data = fgetcsv($handle, 0, ",")) !== FALSE) {
    	$filename = $data[0];
    	$text = $data[1];

    	echo "Creating $filename using say...\n";
    	exec("say --data-format=LEF32@22050 -o $filename \"$text\"");
    }
    fclose($handle);
}
?>
```
Attached the generated sound and updated CSV file as well. 
[audio_pt.zip](https://github.com/dgiese/dustcloud/files/2809516/audio_pt.zip)

Cheers!

dgiese, 2019-04-09:
Hi,
whats the license of OSX say? I mean, can we redistribute files generated by it?

fabiomsouto, 2019-04-10:
Unfortunately I'm not a legalese expert, so I'll wait for someone else to chime in.