dgiese/dustcloud #175: support / root / ssh for v11_001633.pkg and v11_001702.pkg v2 / gen2 firmware

nicoh88, 2018-12-30:
Hello,

`v11_001518.pkg` from june 2018 is the last firmware for the 2nd generation which can be rooted and ssh is activated.

`v11_001633.pkg` and `v11_001702.pkg` can be modified with `firmwarebuilder/imagebuilder.sh` and flash the roborock s50 successfully, **but ssh is not active.**

Please support `v11_001633.pkg` and `v11_001702.pkg`.
Thanks for your work! 👍

nicoh88, 2018-12-30:
It will not be a big deal.

With (quick'n'dirty) ...

```
cat > ./etc/rc.local << EOF
sleep 30
service ssh start
exit 0
EOF
```

... in the `firmwarebuilder/imagebuilder.sh` ssh is enabled, but with password prompt.

cryptomilk, 2018-12-30:
Works just fine for me. I have v001792 running on my robot.

nicoh88, 2018-12-30:
@cryptomilk 

v001792? Typo and v001702? 😕 
Without any changes at `imagebuilder.sh` = from master?

Strange! 😢

closed 2018-12-30 by nicoh88

smar000, 2019-01-13:
HI @nicoh88 

Can you explain what you finally did to get ssh 1702 working? Was it adding the service start command in your earlier post, or did you have stop udisks2? If stopping udisks2, can you tell me exactly where you made these changes.

Many thanks

nicoh88, 2019-01-13:
I did not have to customize the `rc.local`.

Problem was the `udisk2` service on the system where I created the firmware image (`imagebuilder.sh`). Because the `udisk2` service mount the image container, while `imagebuilder.sh` was working.

smar000, 2019-01-14:
Ok thanks for the update.