dgiese/dustcloud #188: How to set a maximum charging level

Algram, 2019-01-27:
Hi there,
I would like to set a maximum level for the robot to charge to, for example 80% instead of 100%. This would significantly improve the battery wear and for smaller rooms that would be ideal.

Has anyone here ever done that? If yes I would be very grateful for some suggestions on how to do it with a rooted vacuum.

dgiese, 2019-04-09:
Afaik there is no way to do that, as it is controlled by the STM32 and the battery management unit.

closed 2019-04-09 by dgiese

Ph1r3ph0x, 2019-07-22:
I think it should be possible to patch the STM32 image which can be found in /usr/rockrobo/firmware/img/
We know that the BMU (MAX8731A) is connected via smbus and has the address 0x12. The VoltageControl register of the BMU is 0x15. These values + the voltage that is beeing set should be hardcoded. I'll have a go at disassembling the image and see if I can find something useful...
The biggest question is how to get the image flashed afterwards, I think :D

Algram, 2019-07-22:
Once you have something flashable let me know. I (or better the robot) can be the guinea pig if you need someone to try this. I have a soldering microscope, so I should be able to solder cables anywhere.

Ph1r3ph0x, 2019-07-23:
So I had some success :D 
I had a go with Ghidra but that didn't give me good results. I was able to find a function that prints out battery related error information via some form of communication but wasn't able to find anything that sends register data to the bms. Having worked with the stm32 and the hal library I know that the library functions can be very very cascaded. Therefore finding the correct function in reconstructed c seems rather impossible for me.
I did however find the 0x41A0 value which occurs only once in the entire binary. This would correspond to the BMS register value for charging to 16.8 V. I patched the value to 0x3F50 which corresponds to 16.2 V (~80% charged). I attached the file.
Are there any debug ports exposed on the pcb? Do you know if there is a way to flash the stm32 via the allwinner? I think flashing this could very well brick the robot :/
The most elegant solution would be to add a function to the stm32 firmware which sets a charging voltage below 1.024V on command of the allwinner. This would then stop the charging and would thus enable us to set any desired value in e.g. valetudo. This is however far from what I am capable of doing xD

[RR_RUBY-S_MP_STM32.A3.G1_M3.3.0_RELEASE_20190403-195148_FULL.zip](https://github.com/dgiese/dustcloud/files/3421928/RR_RUBY-S_MP_STM32.A3.G1_M3.3.0_RELEASE_20190403-195148_FULL.zip)

dgiese, 2019-07-23:
The Allwinner has a SWD connection to the STM32. And the STM32 firmware gets updated also from the software itself

Ph1r3ph0x, 2019-07-23:
So I tried to flash via uart_test. It complains because of a md5 mismatch -.- Any idea, where the hash is beeing stored for comparison? In the file itself? Don't know where to go from here.

Ph1r3ph0x, 2019-07-30:
Just found out that there is a function in "RoboController" called StopCharging()
```
/* CRoboController::StopCharging() */

void StopCharging(void)

{
  CRoboController *this;
  undefined **local_18;
  undefined4 local_14;
  undefined4 local_10;
  int local_c;
  
  local_18 = &PTR_~CRoboEvent+1_00042bfc;
  local_10 = 0;
  local_c = __stack_chk_guard;
  local_14 = 0x10e;
  this = (CRoboController *)GetInstance();
  SendEvent(this,4,(CRoboEvent *)&local_18,true);
  if (local_c == __stack_chk_guard) {
    return;
  }
                    /* WARNING: Subroutine does not return */
  __stack_chk_fail();
}
```
Even if it is not possible to call this function this means that the stm32 is able to stop the charging process on command. It might be possible to use this somehow...